ha-puzzles/evcc-grafana-dashboards #1: Script funktioniert bei mir nicht

MilkoKA, 2024-04-10:
Hallo,
ich bekomme beim ausführen des Scriptes keine Daten zurück
vielleicht hast du ein Tip woran das liegen kann.

Infux in einem Dockercontainer in dem auch das script ausgeführt wird.

Änderung im Script:
#!/bin/bash

# Aggregate date into daily chunks

#consts
INFLUXDB="evcc" # Name of the Influx DB, where you write the EVCC data into
INFLUX_USER="grafana" # Your user name. Empty, if no user is required.
INFLUX_PASSWORD="none" # can be anything except an empty string in case no password is set

folgende Ergebnisse bekomme ich:

./evcc-influx-aggregate.sh --today
Aggregating today
2024-04-10: Aggregating daily metrics.
INFO: Query for daily aggregation of measurement chargePower did not return any results.
INFO: Query for daily aggregation of measurement gridPower did not return any results.
INFO: Query for daily aggregation of measurement batteryPower did not return any results.
INFO: Query for daily aggregation of measurement batteryPower did not return any results.
2024-04:    Aggregating monthly metrics.
invalid timestamp string
INFO: Query for monthly aggregation of measurement pvDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement homeDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement carDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement garageDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement stellplatzDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement gridDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement feedInDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement dischargeDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement chargeDailyEnergy did not return any results.

Zugriff auf die Datenbank scheint zu funktionieren:

influx -database evcc -execute 'show measurements'
name: measurements
name
----
bufferSoc
bufferStartSoc
carDailyEnergy
chargeCurrent
chargePower
chargeRemainingEnergy
chargedEnergy
chargerIcon
chargerPhysicalPhases
disableThreshold
effectiveLimitSoc
effectiveMaxCurrent
effectiveMinCurrent
effectivePlanSoc
effectivePriority
enableThreshold
fatal
feedInDailyEnergy
garageDailyEnergy
greenShareHome
greenShareLoadpoints
gridDailyEnergy
gridPower
homeDailyEnergy
homePower
limitEnergy
limitSoc
maxCurrent
minCurrent
phasesActive
phasesConfigured
phasesEnabled
planEnergy
priority
prioritySoc
pvDailyEnergy
pvEnergy
pvPower
residualPower
sessionCo2PerKWh
sessionEnergy
sessionPrice
sessionPricePerKWh
sessionSolarPercentage
tariffCo2
tariffCo2Home
tariffCo2Loadpoints
tariffGrid
tariffPriceHome
tariffPriceLoadpoints
vehicleClimaterActive
vehicleLimitSoc
vehicleOdometer
vehicleRange
vehicleSoc
vehicleTargetSoc

Danke schonmal im Vorfeld
Gruß
Milko

cschlipf, 2024-04-10:
Welche Influx Version hast Du?

Der entscheidende Punkt ist Zeile 133 im Script: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/35279cadf5f8c97d99a9d51bf9731df7d454cca8/scripts/evcc-influx-aggregate.sh#L133 (dasselbe in Zeile 179, aber lass uns erstmal auf Zeile 133 konzentrieren).

Die generierte Query scheint hier keine Ergebnisse zu liefern

Bitte füge in die Zeile danach mal folgendes ein:
```
echo "$query"
echo "$queryResult"
```

Schick mir bitte den Output.

cschlipf, 2024-04-10:
Was mir auch auffällt: Du hast oben einen Usernamen für Influx angegeben - unten wo Du die measurements abfragst aber nicht. Was passiert wenn Du den Usernamen im Script leer lässt?

MilkoKA, 2024-04-10:
Hi,
sorry, dass ich jetzt erst zum testen komme, da ich noch arbeite.

Mit und ohne Benutzername das gleiche
Influx Version = 1.8.10
./evcc-influx-aggregate.sh  --today

`2024-04-10: Aggregating daily metrics.
SELECT integral("value") / 3600 FROM pvPower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 GROUP BY time(1d) fill(none)
2024-04-10T00:00:00Z 4816.906512361105
SELECT integral("value") / 3600 FROM homePower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 GROUP BY time(1d) fill(none)
2024-04-10T00:00:00Z 8240.045708611116
SELECT integral("value") / 3600 FROM chargePower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 GROUP BY time(1d) fill(none)
2024-04-10T00:00:00Z 0
SELECT integral("value") / 3600 FROM chargePower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND (loadpoint::tag = 'Garage') AND value < 20000 GROUP BY time(1d) fill(none)
2024-04-10T00:00:00Z 0
SELECT integral("value") / 3600 FROM chargePower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND (loadpoint::tag = 'Stellplatz') AND value < 20000 GROUP BY time(1d) fill(none)

INFO: Query for daily aggregation of measurement chargePower did not return any results.
SELECT integral("subquery") / 3600 FROM (SELECT mean("value") AS "subquery" FROM gridPower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 AND "value" >=0 GROUP BY time(10s) fill(none)) WHERE time > '2024-04-10T00:00:00Z' AND time < '2024-04-10T23:59:59Z' GROUP BY time(1d) fill(null)
2024-04-10T00:00:00Z 3937.468722222222
SELECT integral("subquery") / -3600 FROM (SELECT mean("value") AS "subquery" FROM gridPower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 AND "value" <=0 GROUP BY time(10s) fill(none)) WHERE time > '2024-04-10T00:00:00Z' AND time < '2024-04-10T23:59:59Z' GROUP BY time(1d) fill(null)
2024-04-10T00:00:00Z 6833.4259999999995
SELECT integral("subquery") / 3600 FROM (SELECT mean("value") AS "subquery" FROM batteryPower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 AND "value" >=0 GROUP BY time(10s) fill(none)) WHERE time > '2024-04-10T00:00:00Z' AND time < '2024-04-10T23:59:59Z' GROUP BY time(1d) fill(null)

INFO: Query for daily aggregation of measurement batteryPower did not return any results.
SELECT integral("subquery") / -3600 FROM (SELECT mean("value") AS "subquery" FROM batteryPower WHERE time >= '2024-04-10T00:00:00Z' AND time <= '2024-04-10T23:59:59Z' AND value < 20000 AND "value" <=0 GROUP BY time(10s) fill(none)) WHERE time > '2024-04-10T00:00:00Z' AND time < '2024-04-10T23:59:59Z' GROUP BY time(1d) fill(null)

INFO: Query for daily aggregation of measurement batteryPower did not return any results.
2024-04:    Aggregating monthly metrics.
invalid timestamp string
INFO: Query for monthly aggregation of measurement pvDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement homeDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement carDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement garageDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement stellplatzDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement gridDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement feedInDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement dischargeDailyEnergy did not return any results.
invalid timestamp string
INFO: Query for monthly aggregation of measurement chargeDailyEnergy did not return any results.`

ich hoffe das richtig gemacht habe mit "Bitte als Code formatiert."
Gruß Milko

cschlipf, 2024-04-10:
Ich hab's mal umformatiert :-)

Vielen Dank. Ich schau da heute abend mal genauer rein. Es sehe nun aber, dass einige Measurements durchaus Werte zurückgeben. chargePower und batteryPower scheinen jedoch keine Werte zurückzugeben. gridPower dagegen scheint nun Werte zu haben (im ersten Versuch noch nicht). 

Da gridPower wie Battery Power zweimal abgefragt wird: einmal positive Werte für Netzverbrauch und einmal negative Werte für FeedIn vermute ich, dass das gar kein Problem ist: Vermutlich gab es einfach noch keine Einspeisung beim ersten Versuch.

chargePower ist vermutlich auch OK - da Du heute noch kein Auto geladen hast? Ist das korrekt?

Frage nun zur BatteryPower. Hast Du vielleicht keinen Hausspeicher?

Was ich da gerade merke: In den Queries sind bei mir noch meine eigenen Loadpoint Namen Hardcodiert - sogar in die Measurements. Da muss ich nachbessern.

cschlipf, 2024-04-10:
Bei den monthly aggregations scheint es nun noch ein anderes Problem zu geben:

```
invalid timestamp string
```

Kannst Du hier mal diese Zeile auskommentieren?
https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/35279cadf5f8c97d99a9d51bf9731df7d454cca8/scripts/evcc-influx-aggregate.sh#L178

Und dann bitte mir nochmal den Output schicken.

MilkoKA, 2024-04-10:
Hi, 
Auto wurde nicht geladen und Batterie habe ich nicht im Einsatz 
```

Aggregating today
2024-04-10: Aggregating daily metrics.
INFO: Query for daily aggregation of measurement chargePower did not return any results.
INFO: Query for daily aggregation of measurement batteryPower did not return any results.
INFO: Query for daily aggregation of measurement batteryPower did not return any results.
2024-04:    Aggregating monthly metrics.
SELECT sum("value") FROM pvDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement pvDailyEnergy did not return any results.
SELECT sum("value") FROM homeDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement homeDailyEnergy did not return any results.
SELECT sum("value") FROM carDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement carDailyEnergy did not return any results.
SELECT sum("value") FROM garageDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement garageDailyEnergy did not return any results.
SELECT sum("value") FROM stellplatzDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement stellplatzDailyEnergy did not return any results.
SELECT sum("value") FROM gridDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement gridDailyEnergy did not return any results.
SELECT sum("value") FROM feedInDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement feedInDailyEnergy did not return any results.
SELECT sum("value") FROM dischargeDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement dischargeDailyEnergy did not return any results.
SELECT sum("value") FROM chargeDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-00T23:59:59Z'
invalid timestamp string
INFO: Query for monthly aggregation of measurement chargeDailyEnergy did not return any results.
```

cschlipf, 2024-04-10:
Gut, dann ist das bei den daily Aggregations geklärt. Hier muss ich die Meldungen etwas entschärfen und eine Option einführen, dass ein Heimspeicher auch ignoriert wird.

Bei Monthly scheint das wohl ein Problem mit dem Insert Statement zu sein, nicht mit der Abfrage. Ich plane hier mehr Debug Output einzubauen. Siehe #2

MilkoKA, 2024-04-10:
Ich denke das Problem liegt  an numDays=${DAYS_OF_MONTH[$month]}.
im Query steht dann nicht 30 sondern 00.
SELECT sum("value") FROM feedInDailyEnergy WHERE time >= '2024-04-01T00:00:00Z' AND time <= '2024-04-**00**T23:59:59Z'

cschlipf, 2024-04-10:
Ich kann es reproduzieren und ja, das 00 scheint das Problem zu sein. Aber ich weiß noch nicht warum. Aus $month with mit `printf -v fMonth "%02d" $month` ein zweistelliger Monat gemacht - bei der Abfrage muss sie aber schon zweistellig sein. Führe ich `evcc-influx-aggregate.sh --month 2023 4` aus tut es auch. Der Fehler liegt woanders. Aber ich bin dran.

cschlipf, 2024-04-10:
Problem gefunden. 'date +%m' gibt den Monat immer zweistellig zurück. Hier muss ich ihn erst wieder einstellig machen.

Ist gefixt. Aber vielleicht wartest Du noch kurz - habe nun noch ein paar andere Änderungen im Backlog (siehe andere issues), die ich noch gerne fixen würde, bevor du wieder testest.

closed 2024-04-10 by cschlipf

MilkoKA, 2024-04-10:
Super, ich warte noch bis du die anderen Fixes durchgeführt hast.
Bei deiner Anleitung zum Script in Crontab hätte evtl  ein >>pfad/Logdatei angehängt um auch ein Log zu haben.
Habe ich bei mir so gemacht.

Gruß Milko