ha-puzzles/evcc-grafana-dashboards #14: Aggregation not working on Ubuntu 22.04 with influx 2

cschnatz, 2024-04-15:
`/etc/evcc/evcc-influx-aggregate.sh --today
[INFO]  2024-04-15: Aggregating daily metrics.
Error: flag provided but not defined: -database
[INFO]  There is no data from pvPower for pvDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from homePower for homeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from chargePower for carDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from chargePower for loadpoint1DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from gridPower for gridDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from gridPower for feedInDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from batteryPower for dischargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from batteryPower for chargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  2024-04: Aggregating monthly metrics.
Error: flag provided but not defined: -database
[INFO]  There is no data from pvDailyEnergy for pvMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from homeDailyEnergy for homeMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from carDailyEnergy for carMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from loadpoint1DailyEnergy for loadpoint1MonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from gridDailyEnergy for gridMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from feedInDailyEnergy for feedInMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
c[INFO]  There is no data from dischargeDailyEnergy for dischargeMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.
Error: flag provided but not defined: -database
[INFO]  There is no data from chargeDailyEnergy for chargeMonthlyEnergy. This may be fine, if no data had been collected for this month and this measurement.`

Seams like influx2 does not provide the same CLI Client as before.

`influx version
Influx CLI dev (git: none) build_date: 2023-04-28T14:24:14Z`

Any ideas how to get this working? I'm not very familiar with influx...

cschlipf, 2024-04-16:
Can you please send me the outputs of these commands?

```
influx --version
influx --help
```

cschlipf, 2024-04-16:
Influx 2 really seems to have major changes in the CLI. But there seems to be a backward compatibility layer:

https://docs.influxdata.com/influxdb/v2/reference/cli/influx/v1/

Can you test the folowing: Whenever `influx` is called replace the call by `influx v1`?

Here are the occurences:

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b4d3a3655b6239ca058fa758b6f104c7d93e596d/scripts/evcc-influx-aggregate.sh#L128

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b4d3a3655b6239ca058fa758b6f104c7d93e596d/scripts/evcc-influx-aggregate.sh#L167

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b4d3a3655b6239ca058fa758b6f104c7d93e596d/scripts/evcc-influx-aggregate.sh#L175

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b4d3a3655b6239ca058fa758b6f104c7d93e596d/scripts/evcc-influx-aggregate.sh#L227

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b4d3a3655b6239ca058fa758b6f104c7d93e596d/scripts/evcc-influx-aggregate.sh#L235

Just try to add `v1` after each `influx` command call.

cschnatz, 2024-04-16:
`root@csc-evcc:~# influx version
Influx CLI dev (git: none) build_date: 2023-04-28T14:24:14Z
root@csc-evcc:~# influx --help
NAME:
   influx - Influx Client

USAGE:
   influx [command]

HINT: If you are looking for the InfluxQL shell from 1.x, run "influx v1 shell"

COMMANDS:
   version              Print the influx CLI version
   write                Write points to InfluxDB
   bucket               Bucket management commands
   completion           Generates completion scripts
   query                Execute a Flux query
   config               Config management commands
   org, organization    Organization management commands
   delete               Delete points from InfluxDB
   user                 User management commands
   task                 Task management commands
   telegrafs            List Telegraf configuration(s). Subcommands manage Telegraf configurations.
   dashboards           List Dashboard(s).
   export               Export existing resources as a template
   secret               Secret management commands
   v1                   InfluxDB v1 management commands
   auth, authorization  Authorization management commands
   apply                Apply a template to manage resources
   stacks               List stack(s) and associated templates. Subcommands manage stacks.
   template             Summarize the provided template
   bucket-schema        Bucket schema management commands
   scripts              Scripts management commands
   ping                 Check the InfluxDB /health endpoint
   setup                Setup instance with initial user, org, bucket
   backup               Backup database
   restore              Restores a backup directory to InfluxDB
   remote               Remote connection management commands
   replication          Replication stream management commands
   server-config        Display server config
   help, h              Shows a list of commands or help for one command

GLOBAL OPTIONS:
   --help, -h  show help
`

Looks like only v1 is not enough. Tried with `influx v1 shell` but did not work either. 

`root@csc-evcc:~# influx v1 -database evcc -execute 'show measurements' 
Incorrect Usage. flag provided but not defined: -database

NAME:
   influx v1 - InfluxDB v1 management commands

USAGE:
   influx v1 command [command options] [arguments...]

COMMANDS:
   dbrp                 Commands to manage database and retention policy mappings for v1 APIs
   auth, authorization  Authorization management commands for v1 APIs
   shell                Start an InfluxQL shell

OPTIONS:
   --help, -h  show help
   
Error: flag provided but not defined: -database
root@csc-evcc:~# influx v1 shell -database evcc -execute 'show measurements' 
Incorrect Usage: flag provided but not defined: -database

NAME:
   influx v1 shell - Start an InfluxQL shell

USAGE:
   influx v1 shell [command options] [arguments...]

DESCRIPTION:
   Start an InfluxQL shell

COMMON OPTIONS:
   --host value                     HTTP address of InfluxDB [$INFLUX_HOST]
   --skip-verify                    Skip TLS certificate chain and host name verification [$INFLUX_SKIP_VERIFY]
   --configs-path value             Path to the influx CLI configurations [$INFLUX_CONFIGS_PATH]
   --active-config value, -c value  Config name to use for command [$INFLUX_ACTIVE_CONFIG]
   --http-debug                     
   --token value, -t value          Token to authenticate request [$INFLUX_TOKEN]
   
OPTIONS:
   --org-id value         The ID of the organization [$INFLUX_ORG_ID]
   --org value, -o value  The name of the organization [$INFLUX_ORG]
   
Error: flag provided but not defined: -database`

cschnatz, 2024-04-16:
You can use influx query to get data with a token but i'm not familiar with the query itself so i don't know what to query ;) 

```influx query 'from(bucket:"$INFLUXDB") |> range(start:-1m)' --org "$INFLUX_ORG" -t="$INFLUX_API_TOKEN"```

cschlipf, 2024-04-16:
@cschnatz - this is using Flux, but my dashboards are running with InfluxQL, not with (the unfortunately now soon to be dead) Flux query language.

Thank you very much for all the help. I fear I will either need a contribution here from someone using the Influx 2 by providing an alternative script or better: just alternative calls of the few flux commands. Or (worst case) I need to limit the support to Influx 1.8, which I don't want to do.

cschnatz, 2024-04-17:
You Dashboards are awesome. So maybe someone in the EVCC Community could help with this?

cschlipf, 2024-04-20:
An alternative approach could be: Use tasks under Influx 2 to create the aggregations: https://docs.influxdata.com/influxdb/v2/process-data/get-started/

Probably the cleaner method and does not require a shell script, so it would run under system like HA OS as well, where users cannot execute a shell script.

However tasks require Flux and I am not sure you can mix a task using Flux with a DB that is configured to use InfluxQL as required by my dashboard. So it might require a total rewrite of my dashboards using Flux, which then is already announced to become obsolete :-(

Midyr2, 2024-06-13:
Hello,
I've the same problem.
Has anybody an idea to solve this?
I'm using influxdbV2 and InfluxQL

Thanks

Frank

cschlipf, 2024-06-13:
> Hello, I've the same problem. Has anybody an idea to solve this? I'm using influxdbV2 and InfluxQL
> 
> Thanks
> 
> Frank

The only solution I can offer for now is to downgrade to Influx v1.8.x

Midyr2, 2024-06-13:
That was my first idea, but I haven't found a way to do it

cschlipf, 2024-06-13:
Yeah, I fear 'downgrade' really means 'reinstall' in this case :-(

See also https://community.influxdata.com/t/downgrade-influxdb-from-2-x-to-1-8/31563/2. Which in turn means you will loose existing data in your DB. 

As Influx 2 with Flux looks like a dead end today I won't invest any time in supporting it. See also https://www.influxdata.com/blog/the-plan-for-influxdb-3-0-open-source/

Midyr2, 2024-06-13:
I know, that's the reason why I already mapped flux query to InfluxQL.
But if I have understood correctly, the CLI is the problem in the first place, right?

cschlipf, 2024-06-13:
Yes, exactly. I haven't found a way yet to use the `influx` command with Influx 2 to directly run a InfluxQL query from the command line. Running `influx v1` is not fully backward compatible with the Influx 1 CLI tool.

Midyr2, 2024-06-13:
Too bad.
But thank you, especially for your dashboard.
I must have "auf's falsche Pferd gesetzt" with influxdbV2

cschlipf, 2024-06-13:
You are definitely not the only one. I blame Influx for that major failure :-(

cschlipf, 2025-03-16:
Sorry, there is still no solution for that. So I am closing this issue now. Influx 2 is not supported for now and most likely won't be supported at all. Maybe I will support Influx 3 at some point in time.

closed 2025-03-16 by cschlipf