ha-puzzles/evcc-grafana-dashboards #26: Panel plugin not found:

Michel83, 2024-09-28:
Hi, 

ich bin grade dran Ihre Mega Grafana Arbeit umzusetzen.

Leider habe ich beim Import dass Problem:

Panel plugin not found:

[image]

Vieleicht eine Idee, Version von Grafana ist [Grafana v11.2.0 (2a88694fd3)]

Dann hätte ich noch eine kleine Frage die Influx Datenbank läuft bei mir auf einem Pi3 kann ich dort das Skript auch direkt installieren ?

Wenn ja gibt es eine kleine Anleitung, wo am besten zu installieren ?

Ich hatte vorher Ihre Version ohne das Skript am laufen hier gingen die Panels eigentlich alle.

Grüße

cschlipf, 2024-09-28:
Die Dokumentation zur Installation des Skriptes habe ich gerade aktualisiert: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/main/scripts/README.md

Vielen Dank für das Feedback.

Michel83, 2024-09-28:
Die Plugins habe ich grade verglichen und sind alle gleich

Ein unterschied habe ich hier:

Bibliotheks-Panels

[image]

Wenn ich richtig bin müssten die Panels doch erst mal ohne das Script laufen, also klar ohne daten aber da sein müssten sie.

Ich werfe gleich nochmal alle raus, und importiere neu.

Und danke für die schnelle Antwort, Grüße.

cschlipf, 2024-09-28:
Die Library Panels haben sich in der Zwischenzeit auch geändert. Daher sind hier Unterschiede vollkommen OK. Bin mal gespannt was bei dem neu-import herauskommt. Falls das hilft, werde ich hier den Upgrade Pfad dokumentieren.

Michel83, 2024-09-28:
Hat geklappt, das löschen der Bibliotheks-Panels war die Lösung.

[image]

Jetzt versuche ich mich mal an dem script.

cschlipf, 2024-09-28:
Super! Vielen Dank. Dann werde ich das mal besser dokumentieren.

cschlipf, 2024-09-28:
Dazu musst Du das Script etwas anpassen, aber sollte gut machbar sein.

Hier die Variablen für Loadpoint_2 für Loadpoint_3 duplizieren: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/ce632819a7713fbffa8078098fb86426de312e8e/scripts/evcc-influx-aggregate.sh#L14

Diesen IF Block duplizieren und für Loadpoint_3 anpassen: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/ce632819a7713fbffa8078098fb86426de312e8e/scripts/evcc-influx-aggregate.sh#L189

Und dasselbe noch einmal für the Monat: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/ce632819a7713fbffa8078098fb86426de312e8e/scripts/evcc-influx-aggregate.sh#L247

Das sollte es gewesen sein.

cschlipf, 2024-09-28:
Ich mach den Issue nochmal auf. Will ihn nutzen um die Upgrade Doku anzupassen.

Michel83, 2024-09-28:
So richtig ?

```
`#!/bin/bash

# Aggregate date into daily chunks

#consts
INFLUXDB="evcc" # Name of the Influx DB, where you write the EVCC data into.
INFLUX_USER="" # Your user name. Empty, if no user is required. Default: ""
INFLUX_PASSWORD="none" # can be anything except an empty string in case no password is set.  Default: "none"
INFLUX_HOST="localhost" # If the script is run remotely, enter the host name of the remote host. Default: "localhost"
INFLUX_PORT=8086 # The port to connect to influx. Default: 8086
HOME_BATTERY="true" # Set to false in case your home does not use a battery.
DEBUG="false" # Set to true to generate debug output.
LOADPOINT_1_TITLE="Garage Wallbox" # Title of loadpoint 1 as defined in evcc.yaml
LOADPOINT_2_ENABLED=true # Set to false in case you have just one loadpoint
LOADPOINT_2_TITLE="Werkstatt Niu Charger" # Title of loadpoint 2 as defined in evcc.yaml
LOADPOINT_3_ENABLED=true # Set to false in case you have just one loadpoint 
LOADPOINT_3_TITLE="AC Laden 16A Speicher" # Title of loadpoint 2 as defined in evcc.yaml

#arguments
AGGREGATE_YEAR=0
AGGREGATE_YESTERDAY=false
AGGREGATE_TODAY=false
AGGREGATE_MONTH_YEAR=0
AGGREGATE_MONTH_MONTH=0
AGGREGATE_DAY_YEAR=0
AGGREGATE_DAY_MONTH=0
AGGREGATE_DAY_DAY=0

# Maps of number of days per month. February leap year is updated later
declare -A DAYS_OF_MONTH
DAYS_OF_MONTH[1]=31
DAYS_OF_MONTH[2]=28
DAYS_OF_MONTH[3]=31
DAYS_OF_MONTH[4]=30
DAYS_OF_MONTH[5]=31
DAYS_OF_MONTH[6]=30
DAYS_OF_MONTH[7]=31
DAYS_OF_MONTH[8]=31
DAYS_OF_MONTH[9]=30
DAYS_OF_MONTH[10]=31
DAYS_OF_MONTH[11]=30
DAYS_OF_MONTH[12]=31

logError() {
    echo "[ERROR] $1"
}

logInfo() {
    echo "[INFO]  $1"
}

logDebug() {
    if [ "$DEBUG" == "true" ]; then
        echo "[DEBUG] $1"
    fi
}

parseArguments() {
    if [ "$#" -eq 0 ]; then
        printUsage
        exit 1
    fi
    if [ "$1" == '--year' ]; then
        if [ "$#" -ne 2 ]; then
            printUsage
            exit 1
        fi
[... 235 lines cut ...]
    year=`date -d yesterday +%Y`
    # Converting to a base 10 number, stripping of a leading 0
    month=$(( 10#`date -d yesterday +%m`))
    day=$(( 10#`date -d yesterday +%d`))
    if isLeapYear $year; then
        logDebug "The year $year is a leap year. February has 29 days."
        DAYS_OF_MONTH[2]=29
    fi
    aggregateDay $year $month $day
    aggregateMonth $year $month
elif [ "$AGGREGATE_TODAY" == "true" ]; then
    year=`date +%Y`
    # Converting to a base 10 number, stripping of a leading 0
    month=$(( 10#`date +%m`))
    day=$(( 10#`date +%d`))
    if isLeapYear $year; then
        logDebug "The year $year is a leap year. February has 29 days."
        DAYS_OF_MONTH[2]=29
    fi
    aggregateDay $year $month $day
    aggregateMonth $year $month
fi

### END`
```

cschlipf, 2024-09-28:
Nicht ganz.

Bei aggregateDay():
```bash
    if [ "$LOADPOINT_2_ENABLED" == "true" ]; then
        writeDailyEnergies "all" "value" "chargePower" "loadpoint2DailyEnergy" $ayear $amonth $aday "AND ("loadpoint"::tag = '${LOADPOINT_2_TITLE}') AND value < 20000"
    else
        logDebug "Loadpoint 2 is disabled."
    fi
		
    if [ "$LOADPOINT_3_ENABLED" == "true" ]; then
        writeDailyEnergies "all" "value" "chargePower" "loadpoint3DailyEnergy" $ayear $amonth $aday "AND ("loadpoint"::tag = '${LOADPOINT_3_TITLE}') AND value < 20000"
    else
        logDebug "Loadpoint 3 is disabled."
    fi
```

Bei aggregateMonth():
```bash
     if [ "$LOADPOINT_2_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint2DailyEnergy" "loadpoint2MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 2 is disabled."
    fi

     if [ "$LOADPOINT_3_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint3DailyEnergy" "loadpoint3MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 3 is disabled."
    fi
```

closed 2024-09-28 by cschlipf

Michel83, 2024-09-29:
Hi, habe grade mal mit dem Scrip verssucht:

```
#!/bin/bash

# Aggregate date into daily chunks

#consts
INFLUXDB="evcc" # Name of the Influx DB, where you write the EVCC data into.
INFLUX_USER="" # Your user name. Empty, if no user is required. Default: ""
INFLUX_PASSWORD="none" # can be anything except an empty string in case no password is set.  Default: "none"
INFLUX_HOST="localhost" # If the script is run remotely, enter the host name of the remote host. Default: "localhost"
INFLUX_PORT=8086 # The port to connect to influx. Default: 8086
HOME_BATTERY="true" # Set to false in case your home does not use a battery.
DEBUG="false" # Set to true to generate debug output.
LOADPOINT_1_TITLE="Garage Wallbox" # Title of loadpoint 1 as defined in evcc.yaml
LOADPOINT_2_ENABLED=true # Set to false in case you have just one loadpoint
LOADPOINT_2_TITLE="Werkstatt Niu Charger" # Title of loadpoint 2 as defined in evcc.yaml
LOADPOINT_3_ENABLED=true # Set to false in case you have just one loadpoint 
LOADPOINT_3_TITLE="AC Laden 16A Speicher" # Title of loadpoint 3 as defined in evcc.yaml

#arguments
AGGREGATE_YEAR=0
AGGREGATE_YESTERDAY=false
AGGREGATE_TODAY=false
AGGREGATE_MONTH_YEAR=0
AGGREGATE_MONTH_MONTH=0
AGGREGATE_DAY_YEAR=0
AGGREGATE_DAY_MONTH=0
AGGREGATE_DAY_DAY=0

# Maps of number of days per month. February leap year is updated later
declare -A DAYS_OF_MONTH
DAYS_OF_MONTH[1]=31
DAYS_OF_MONTH[2]=28
DAYS_OF_MONTH[3]=31
DAYS_OF_MONTH[4]=30
DAYS_OF_MONTH[5]=31
DAYS_OF_MONTH[6]=30
DAYS_OF_MONTH[7]=31
DAYS_OF_MONTH[8]=31
DAYS_OF_MONTH[9]=30
DAYS_OF_MONTH[10]=31
DAYS_OF_MONTH[11]=30
DAYS_OF_MONTH[12]=31

logError() {
    echo "[ERROR] $1"
}

logInfo() {
    echo "[INFO]  $1"
}

logDebug() {
    if [ "$DEBUG" == "true" ]; then
        echo "[DEBUG] $1"
    fi
}

parseArguments() {
    if [ "$#" -eq 0 ]; then
        printUsage
        exit 1
    fi
    if [ "$1" == '--year' ]; then
        if [ "$#" -ne 2 ]; then
            printUsage
            exit 1
        fi
[... 252 lines cut ...]
    aggregateDay $year $month $day
    aggregateMonth $year $month
fi

### END
```

Problem er findet keine Daten:
```
There is no data from chargePower for loadpoint1DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from chargePower for loadpoint2DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from chargePower for loadpoint3DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from batteryPower for dischargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from batteryPower for chargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  2024-05-26: Aggregating daily metrics.
[INFO]  There is no data from chargePower for loadpoint1DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from chargePower for loadpoint2DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from chargePower for loadpoint3DailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from batteryPower for dischargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  There is no data from batteryPower for chargeDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
[INFO]  2024-05-27: Aggregating daily metrics.
```
Gibt ess vieleicht noch ein Tipp ?

Grüße

Michel83, 2024-09-29:
Erledigt, es Funktioniert. Danke :-)

Michel83, 2024-09-29:
```
# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command
5 0 * * * home/pi/evcc-influx-aggregate.sh --yesterday >> /var/log/evcc-grafana-dashboards.log 2>&1
0 * * * * home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&
```

Da müsste ja so dann passen.

Hier liegt der Script:
[image]

PR #29 (2024-09-28): Dokumentation Upgrade Pfad
Close #26
commit dd622adc (2024-09-28): Dokumentation Upgrade Pfad (#29)

Close #26