ha-puzzles/evcc-grafana-dashboards #31: crontab funktioniert nicht

Michel83, 2024-10-03:
Hallo,

leider funktioniert bei mir der Aufruf nicht, ich habe jetzt schon mal den Log eingeschaltet aber so wirklich schlau werde ich daraus nicht. Siehe unten:

```
Oct  3 05:00:01 Influxdb1 CRON[24196]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 05:00:01 Influxdb1 CRON[24195]: (CRON) info (No MTA installed, discarding output)
Oct  3 05:17:01 Influxdb1 CRON[3111]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct  3 06:00:01 Influxdb1 CRON[5035]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 06:00:01 Influxdb1 CRON[5034]: (CRON) info (No MTA installed, discarding output)
Oct  3 06:17:01 Influxdb1 CRON[5838]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct  3 06:25:01 Influxdb1 CRON[6231]: (root) CMD (test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.daily ))
Oct  3 07:00:01 Influxdb1 CRON[7851]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 07:00:01 Influxdb1 CRON[7850]: (CRON) info (No MTA installed, discarding output)
Oct  3 07:17:01 Influxdb1 CRON[8653]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct  3 08:00:01 Influxdb1 CRON[10600]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 08:00:01 Influxdb1 CRON[10599]: (CRON) info (No MTA installed, discarding output)
Oct  3 08:17:01 Influxdb1 CRON[11386]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct  3 09:00:01 Influxdb1 CRON[13307]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 09:00:01 Influxdb1 CRON[13306]: (CRON) info (No MTA installed, discarding output)
Oct  3 09:17:01 Influxdb1 CRON[14090]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct  3 10:00:01 Influxdb1 CRON[16045]: (pi) CMD (root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&)
Oct  3 10:00:01 Influxdb1 CRON[16044]: (CRON) info (No MTA installed, discarding output)
```

Der Eintrag:

```
# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command
5 0 * * * root /home/pi/evcc-influx-aggregate.sh --yesterday >> /var/log/evcc-grafana-dashboards.log 2>&1
0 * * * * root /home/pi/evcc-influx-aggregate.sh --today >> /var/log/evcc-grafana-dashboards.log 2>&

```

Grüße

cschlipf, 2024-10-03:
Läuft das script, wenn Du "/home/pi/evcc-influx-aggregate.sh --yesterday" auf der Kommandozeile ausführst?

Ich kann Dir hier leider keinen Support für Crontab oder Dein Linuxsystem geben.

Michel83, 2024-10-04:
Hi danke erst mal für den super Support! 

Also der Script läuft wenn ich ihn in der Kommandozeile ausführe, habe jetzt mal den root weglassen und teste gefunden hatte ich das in einem anderen forum.

cschlipf, 2024-10-04:
Du editierst das schon mit `crontab -e`, oder?

Michel83, 2024-10-04:
Ja, anders geht ja auch glaube ich garnicht.

Michel83, 2024-10-05:
So wie es aussieht funktioniert es jetzt es war ein Schreibrecht Problem, die Datei `evcc-grafana-dashboards.log` erstellt die Rechte angepasst und siehe da es steht dann auch was drinnen.

Hätte der crontab log ja gesagt das er keine schreibrechte hatte wäre es wahrscheinlich einfacher gewesen. 

Vielleicht kann man das ja noch in deine Beschreibung aufnehmen, wenn nochmal jemand darüber stolpern sollte.

PR #33 (2024-10-06): Document how to create the script log file and set permissions
Close #31
closed 2024-10-06 by cschlipf