ha-puzzles/evcc-grafana-dashboards #34: Fehlednde Datenpunkte

Michel83, 2024-10-07:
Hi,

ich muss schon wieder nerven.

Ich habe grade festgestellt das mir ein Datenpunkte fehlt:

[image]

[image]

gibt es eine Möglichkeit diese zu überprüfen oder neu anlegen zu lassen.

Grüße

cschlipf, 2024-10-07:
Sieht so aus, als würde es ein Problem in Deinem angepassten Script in der Funktion aggregateMonth() geben. Alle Aggregationen, die nach den Loadpoints kommen, fehlen hier. Also alles ab dieser Zeile:

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b01adf2ca45a4d6a44a9d2f28d9fa42f3131f6b6/scripts/evcc-influx-aggregate.sh#L252

Kannst Du mal Deine angepasste `aggregateMonth()` Funktion hier posten? Bitte als Code formatiert.

Michel83, 2024-10-07:
Dein Support ist echt der Hammer !

```
aggregateMonth() {
    ayear=$1
    amonth=$2

    logInfo "`printf "%04d" $ayear`-`printf "%02d" $amonth`: Aggregating monthly metrics."

    writeMonthlyEnergies "value" "pvDailyEnergy" "pvMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "homeDailyEnergy" "homeMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "carDailyEnergy" "carMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "loadpoint1DailyEnergy" "loadpoint1MonthlyEnergy" $ayear $amonth 
        if [ "$LOADPOINT_2_ENABLED" == "true" ]; then
    writeMonthlyEnergies "value" "loadpoint2DailyEnergy" "loadpoint2MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 2 is disabled."
    fi

    if [ "$LOADPOINT_3_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint3DailyEnergy" "loadpoint3MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 3 is disabled."
    fi
}

###############################################################################
### MAIN
###############################################################################
parseArguments $@
```

cschlipf, 2024-10-07:
Joah... das ist dann klar. Du hast hier den Rest der Funktion entfernt ;-).

Füge mal wieder die Zeilen 252-260 ein: https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/b01adf2ca45a4d6a44a9d2f28d9fa42f3131f6b6/scripts/evcc-influx-aggregate.sh#L252-L260

closed 2024-10-07 by cschlipf

Michel83, 2024-10-07:
wie auch immer das passieren konnte.......

```
aggregateMonth() {
    ayear=$1
    amonth=$2

    logInfo "`printf "%04d" $ayear`-`printf "%02d" $amonth`: Aggregating monthly metrics."

    writeMonthlyEnergies "value" "pvDailyEnergy" "pvMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "homeDailyEnergy" "homeMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "carDailyEnergy" "carMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "loadpoint1DailyEnergy" "loadpoint1MonthlyEnergy" $ayear $amonth
        if [ "$LOADPOINT_2_ENABLED" == "true" ]; then
    writeMonthlyEnergies "value" "loadpoint2DailyEnergy" "loadpoint2MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 2 is disabled."
    fi

    if [ "$LOADPOINT_3_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint3DailyEnergy" "loadpoint3MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 3 is disabled."

    fi

        writeMonthlyEnergies "value" "gridDailyEnergy" "gridMonthlyEnergy" $ayear $amonth
        writeMonthlyEnergies "value" "feedInDailyEnergy" "feedInMonthlyEnergy" $ayear $amonth

    if [ "$HOME_BATTERY" == "true" ]; then
        writeMonthlyEnergies "value" "dischargeDailyEnergy" "dischargeMonthlyEnergy" $ayear $amonth
        writeMonthlyEnergies "value" "chargeDailyEnergy" "chargeMonthlyEnergy" $ayear $amonth
    else
        logDebug "Home battery aggregation is disabled"
    fi
}
```

cschlipf, 2024-10-07:
Hab mir mal erlaubt die Einrückungen zu korrigieren. Mir rollen sich sonst die Fußnägel auf ;-)

```bash
aggregateMonth() {
    ayear=$1
    amonth=$2

    logInfo "`printf "%04d" $ayear`-`printf "%02d" $amonth`: Aggregating monthly metrics."

    writeMonthlyEnergies "value" "pvDailyEnergy" "pvMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "homeDailyEnergy" "homeMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "carDailyEnergy" "carMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "loadpoint1DailyEnergy" "loadpoint1MonthlyEnergy" $ayear $amonth

    if [ "$LOADPOINT_2_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint2DailyEnergy" "loadpoint2MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 2 is disabled."
    fi
    if [ "$LOADPOINT_3_ENABLED" == "true" ]; then
        writeMonthlyEnergies "value" "loadpoint3DailyEnergy" "loadpoint3MonthlyEnergy" $ayear $amonth
    else
        logDebug "Loadpoint 3 is disabled."
    fi

    writeMonthlyEnergies "value" "gridDailyEnergy" "gridMonthlyEnergy" $ayear $amonth
    writeMonthlyEnergies "value" "feedInDailyEnergy" "feedInMonthlyEnergy" $ayear $amonth

    if [ "$HOME_BATTERY" == "true" ]; then
        writeMonthlyEnergies "value" "dischargeDailyEnergy" "dischargeMonthlyEnergy" $ayear $amonth
        writeMonthlyEnergies "value" "chargeDailyEnergy" "chargeMonthlyEnergy" $ayear $amonth
    else
        logDebug "Home battery aggregation is disabled"
    fi
}
```

Michel83, 2024-10-07:
Kein Problem, läuft grade durch dauert gefühlt zumindest schon mal länger.

Michel83, 2024-10-07:
Alles Funktioniert, Danke.