ha-puzzles/evcc-grafana-dashboards #58: Vehicle1name is missing as mapping override in Today Dashboard

endurance1968, 2025-01-03:
[image]

endurance1968, 2025-01-03:
Ich habe das gerade nochmals probiert mit def. alle panels gelöscht und import nach server start - das mapping hat wieder gefehlt. Aber auch nur das eine für vehicle1 vehicle2 war drin.
Und ja im JSON ist es da. Keine Ahnung woran es liegt evtl. an verschiedenen Grafan Versionen? Ich nutz 11.4.0

cschlipf, 2025-01-03:
Ich lasse den Issue mal repräsentativ für das Upgrade Problem stehen. Entweder muss ich die Dokumentation erweitern oder andere Lösungen finden. Aber so ist das unbefriedigend.

endurance1968, 2025-01-04:
Moin, speziell zu dem Mapping habe ich die JSON nochmals angeschaut - das Mapping fehlt tatsächlich.
Fehlt:
```
{
              "matcher": {
                "id": "byName",
                "options": "vehicle1"
              },
              "properties": [
                {
                  "id": "displayName",
                  "value": "$vehicle1Name"
                },
                {
                  "id": "color",
                  "value": {
                    "fixedColor": "light-red",
                    "mode": "fixed"
                  }
                }
              ]
```

cschlipf, 2025-01-04:
Ui ui... bei mir fehlt da das gesamte Override. Vielen Dank!

cschlipf, 2025-01-04:
Um sicher zu gehen. Wir reden von diesem Panel, richtig?

[image]

endurance1968, 2025-01-04:
Ja genau. Das Energiepanel - zukünftig schreibe ich den Panelnamen mit dazu.

PR #67 (2025-01-04): Fix Issues #58, #62, #64, and #65
Closes #58
Closes #62 (requires re-aggegration)
Closes #64
Closes #65
commit 302e9117 (2025-01-04): Fix Issues #58, #62, #64, and #65 (#67)

Closes #58
Closes #62
Closes #64
Closes #65
closed 2025-01-04 by cschlipf