ha-puzzles/evcc-grafana-dashboards #59: Dashboard Daily - energie/15min

endurance1968, 2025-01-03:
Namensmappings sollten hinzugefügt werden
[image]
Ansonsten werden autogenerierte Namen angezeigt.

cschlipf, 2025-01-03:
Habe ich eigentlich gemacht. Wo habe ich das verpasst?

endurance1968, 2025-01-03:
Bei mir wurde der Eigenverbrauch als "Eigenverbrauch pvPower.integral" angezeigt.
Hab das wie im im Screenshot hinzugefügt und explizit auf nur Eigenverbrauch gesetzt.

cschlipf, 2025-01-03:
Ist eigentlich schon gesetzt:

[image]

Auch hier vermute ich, dass es daran liegt, dass Du vor dem Import die Library Panels nicht gelöscht hast?

endurance1968, 2025-01-03:
Ich kommentier das gerne auch hier 100% gelöscht - hab mich noch drüber geärgert, dass man die einzeln löschen muss.

cschlipf, 2025-01-03:
Leider nein. Grafana ist hier wirklich schlecht. Die Library Panel können nur mit den Dashboards importiert werden. Werden aber leider nicht neu importiert, wenn sie schon da sind :-(

Siehe auch: https://github.com/ha-puzzles/evcc-grafana-dashboards?tab=readme-ov-file#upgrade

endurance1968, 2025-01-03:
[image]
Also zumindest meine Grafana Version hat ein Problem mit dem Import und Mixed sourcen. Die Quellen Fehlen nach dem Import (und Grafana neustart). Schlimmer noch wenn ich die Quelle Auswähle wird die Query gelöscht. Gut kann man vorher sichern. Aber witzigerweise hat auch die Expression eine source.

cschlipf, 2025-01-03:
Args! Das ist echt Mist :-(

Ich sollte wohl Mixed Data Sources vermeiden und alles was ich im Monthly/Yearly Dashboard brauche in Aggregationen speichern. Wäre eh sinnvoller.

endurance1968, 2025-01-03:
Also ich habe jetzt alle Panel wieder korrekt am laufen - der Teufel liegt evtl. im Detail. Würde mich nicht wundern, wenn das bei anderen mit dem Import wunderbar klappt. Aus meiner sicht kannst du die Tickets schließen und warten ob andere ähnliche Probleme haben.
Wie ich schon schrieb habe ich die JSON importiert und bin nicht über die Grafana IDs gegangen - sollte keinen Unterschied machen - aber das werde ich beim nächsten mal testen. Sind die auf dem Grafana Server dann mit jedem Release auch gleich aktualisiert?

cschlipf, 2025-01-03:
Vielen Dank für das Feedback.

Den Store aktualisiere ich mit dem Release. Die JSON Dateien könnten Zwischenschritte enthalten bis zum nächsten Release.

cschlipf, 2025-01-03:
Closed: Upgrade issue

closed 2025-01-03 by cschlipf