ha-puzzles/evcc-grafana-dashboards #60: monthly dashboard mixed datasource in Speicher panel

endurance1968, 2025-01-03:
die panel queries wurden nicht importiert. vermute mal weil zwei verschiedene Datenquellen drin vorkommen?

Wenn ich ins JSON schaue steht da auch
"datasource": {
          "type": "datasource",
          "uid": "**-- Mixed --**"
        },

cschlipf, 2025-01-03:
Ja genau: 
[image]

Das ist auch ein Library Panel. Hast Du hier auch die Library Panels vor dem erneuten Import gelöscht?

endurance1968, 2025-01-03:
>Das ist auch ein Library Panel. Hast Du hier auch die Library Panels vor dem erneuten Import gelöscht?

100% wie aber im anderen Ticket geschrieben - den Grafana Server habe ich nicht neu gestartet. Habe den Eindruck da bleiben ab und zu Leichen im Speicher die dann Probleme machen.

cschlipf, 2025-01-03:
OK... das ist interessant. Leider habe ich hier nur eine Grafana-Instanz auf der ich entwickle. Daher durchlaufe ich den Upgrade Prozess nicht jedes Mal. Feedback hier ist wichtig für mich.

cschlipf, 2025-01-03:
Closed: Upgrade issue

closed 2025-01-04 by cschlipf