ha-puzzles/evcc-grafana-dashboards #61: Dasboard yearly - loadpointXName fehlt im Energie/Monats Panel

endurance1968, 2025-01-03:
Das Displayname mapping fehlt

Hier von mir schon hinzugefügt
[image]

cschlipf, 2025-01-03:
Kannst Du bitte einen Dashboard Screenshot mit dem problematischen Panel posten? Ich suche das gerade und finde es nicht.

BTW: Mir fällt auf, dass das Field bei Dir noch 'Garage' heisst. Das sollte nun eigentlich 'loadpoint2' heißen. Kann es sein, dass Du die LIbrary Panels vor dem erneuten Import nicht gelöscht hast?

endurance1968, 2025-01-03:
Hatte alles gelöscht zu 100% - evtl. hätte ich Grafana noch neu starten sollen? Ist die Software selbst nach der langen Lebenszeit immer noch so buggy :( (auch wenn sie geil ist)
[image]
da stand bei mir noch Garage und weiss nicht mehr was

cschlipf, 2025-01-03:
Manchmal hilft auch ein Click auf Reload bei gedrückter Shift Taste, weil einiges im Browser Cache drin steckt.

cschlipf, 2025-01-03:
Closed: Upgrade issue

closed 2025-01-03 by cschlipf