ha-puzzles/evcc-grafana-dashboards #63: Beobachtung: Dashbard Today - query format recognized as flux instead of influxQL

endurance1968, 2025-01-04:
Mal nur als Beobachtung im EVCC: PV Verbrauchsverlauf Panel

[image]

Es funktioniert alles von daher ist mir das gestern nicht aufgefallen. Nur kann man die Query so nicht mehr komplett editieren (e.g. das alias fehlt).

Ich habe mal testhalber die PV Leistungs query neu angelegt und dann JSON export verglichen:

Bei Dir:
```
        {
            "alias": "PV",
            "datasource": {
              "type": "influxdb",
              "uid": "L-nZgczgk"
            },
            "groupBy": [
              {
                "params": [
                  "$__interval"
                ],
                "type": "time"
              },
              {
                "params": [
                  "null"
                ],
                "type": "fill"
              }
            ],
            "hide": false,
            "orderByTime": "ASC",
            "policy": "default",
            "query": "SELECT max(\"value\") FROM \"pvPower\" WHERE (\"value\"::field < $peakPowerLimit) AND $timeFilter GROUP BY time(1m) fill(none)  tz('$timezone')",
            "rawQuery": true,
            "refId": "PV Leistung",
            "resultFormat": "time_series",
            "select": [
              [
                {
                  "params": [
                    "value"
                  ],
                  "type": "field"
                },
                {
                  "params": [],
                  "type": "mean"
                }
              ]
            ],
            "tags": []
          },
```

bei mir
```
        {
            "alias": "PV",
            "datasource": {
              "type": "influxdb",
              "uid": "de89y1o57aqkgd"
            },
            "hide": false,
            "query": "SELECT max(\"value\") FROM \"pvPower\" WHERE (\"value\"::field < $peakPowerLimit) AND $timeFilter GROUP BY time(1m) fill(none)  tz('$timezone')",
            "rawQuery": true,
            "refId": "PV Leistung",
            "resultFormat": "time_series"
          }
```
d.h. den extra select und groupBy Abschnitt habe ich nicht. Gut das kommt wohl nur, wenn man die query mit dem Editor zusammenklickt. Wenn ich das mache, sieht es fast so aus wie deins ABER
            "rawQuery": **false**,
Ich hab den Eindruck, dass wenn man mal im raw mode was ändert, der urspüngliche query input erhalten bleibt und dann teils datenmüll erzeugt. Das kommt im Prinzip auch als Warnung, wenn man hin und her schaltet.
Wäre jetzt die Frage, ob der Import reibungsloser funktioniert, wenn man auf saubere raw queries umstellt und auf den editor "balast" verzichtet. 

Bei dem Gauges Panel sieht es gut aus - obwohl auf den ersten Blick das JSON ähnlich ist.

endurance1968, 2025-01-04:
Ich werde bei mir mal weitere Grafana Testserver aufsetzen. Infra kann ich problemlos zum Testen erweitern.
So wie ich das gesehen habe nutzt Du auch Grafana 11.4 also werde ich mal einen frischen Server damit erstellen um damit imports testen ohne das die Ergebnisse von alten Daten verfälscht werden könnten.

endurance1968, 2025-01-04:
Also ich habe die reproduzierbare Ursache für das falsche Erkennen der InfluxQL gefunden. Sobald die default DataSource eine Flux Datasource ist sind rawQuery als Flux dargestellt. Stelle ich auf EVCC (influxQL) als default um ist es InfluxQL. 
Das scheint ein Bug in Grafana zu sein, statt die Sprache der zugeweisenen Source nimmt er die vom Defaultsource.

cschlipf, 2025-01-04:
OK, das kommt bei mir nicht vor, da ich keine Flux Datasources habe. Bei mir ist alles influxQL. Und ja, auch Grafana 11.4.0 bei mir

Vielen Dank, dass Du da dran bleibst und das gefunden hast. Da ich nun aber hier nichts fixen kann und wir auf Grafana warten müssen, mache ich mal hier zu.

closed 2025-01-04 by cschlipf