ha-puzzles/evcc-grafana-dashboards #66: Featurerequest: Show (dyn)Tariff in Daily Dashboard

endurance1968, 2025-01-04:
[image]

SELECT mean("value")  FROM "tariffGrid" WHERE $timeFilter GROUP BY time($__interval) fill(null) tz('$timezone')

evtl. auch view disabled für den Graph als default. Denke für Leute mit dynamischen Strompreis ist das interessant. Also ich schau da immer drauf um zu sehen wann mein Speicher vom Netz nachlädt.
Was ich noch nicht kapiert habe ist, wie man den Graph von 23-00Uhr fertig zeichnet wenn der Tag rum ist.

Wenn man einen festen Preis hat ist das nat. irrelevant.

cschlipf, 2025-01-04:
Siehe Pull Request #68. Ich habe es etwas anders gemacht. Statt im großen Chart mit einzubauen, habe ich ein extra Chart gemacht. Das hat den Vorteil, dass ich teure und günstige Preise mit den Gradienten highlighten kann und außerdem konnte ich so noch den Energieverbrauch zu dieser Kostenstunde mit als Overlay hinzufügen, damit man sieht ob man auch die teuersten Stunden gut vermieden hat.

Außerdem gibt es noch tägliche Aggregationen für das monatliche und jährliche Dashboard.

Vielen Dank für die Idee.

closed 2025-01-04 by cschlipf

endurance1968, 2025-01-04:
Passt - das dann gleich in die anderen Dashboards mit reinzupacken ist super! Danke
Ich warte dann mal auf das nächste Release und würde dann nochmals durchtesten. Diesmal weiss ich ja auch schon wo evtl. Fallstricke sind um sich selbst reinzulegen :)

cschlipf, 2025-01-04:
Vielen Dank. Mit dem nächsten Release warte ich noch ein paar Tage um noch Fehler zu entdecken (just gerade einen gefunden - falsche Preise im Monat und Jahr)

PR #68 (2025-01-04): Fix #66: Show dynamic tariffs
Closes #66

# Today

[image]

Price with a hourly energy consumption overlay

# Monthly

[image]

Maximum, average, and minimum daily prices

# Yearly

[image]

Maximum, average, and minimum daily prices
commit d7f9c351 (2025-01-04): Fix #66: Show dynamic tariffs (#68)

* Fix #66: Show dynamic tariffs

Closes #66

* Added dashboards with tariff widgets