ha-puzzles/evcc-grafana-dashboards #80: Improvemnent request - Dashboard Today Dynamischer Preis

endurance1968, 2025-01-27:
Im Dashboard today

[image]

meist OK aber ab und zu scheint Ihn irgendwelche Einträge durcheinander zu bringen. Passiert wohl vermehrt wenn man evcc mal öfters an einem Tag neu gestartet hat (so kommt es mir zumindest vor).
Evtl. kann man da etwas mit filtern oder sonstiges machen - keine Ahnung wie es bei Dir aussieht  - immer sauber?

cschlipf, 2025-01-31:
Kann ich bei mir leider nicht reproduzieren. Bräuchte da mehr Details.

endurance1968, 2025-01-31:
Ob das bei dir reproduzierbar ist mach ich ein Fragezeichen dran.
Gerade mal geschaut: am Mittwoch evcc neu gestartet:

[image]
gestern durchgelaufen:
[image]
heute war bisher gut, dann habe ich evcc gerade neu gestartet und dann sieht es so aus:
[image]

gibt es tools mit denen man sich die Datensätze in influxdb v1.x einfach anschauen kann - so etwas wie ein browser durch die measurements...? Ich habe den Eindruck da wird bei neustarten irgendetwas reingeschrieben was dann Grafana durcheinander bringt.

cschlipf, 2025-01-31:
Das tool das ich nehme ist das influx CLI. Da kannst Du dann die Measurements per Query abfragen

endurance1968, 2025-01-31:
>Das tool das ich nehme ist das influx CLI. 

OK das kenne ich natürlich, hatte gehofft es gibt was mit klickibunti :), aber mit Gui habe ich nix vernünftiges gefunden - außer dann doch wieder queries eintippen. Dann kann ich auch in Grafana die tabellenansicht nehmen. Und stimmt da sind Einträge mit Zeitstempel vom restart evcc drin.
[image]
kann man evtl. Einträge mit Minute != 0 rausfiltern?

cschlipf, 2025-01-31:
Ah! "Restart" ist ein Schluesselwort hier. Werde gleich mal versuchen, ob ich es damit reproduzieren kann.

cschlipf, 2025-01-31:
Also mit einem einfachern Restart kann ich es nicht reproduzieren. Kann es sein, dass bei Dir EVCC über die Stundengrenze hinweg down war und Du dann danach einen Restart gemacht hast?

Eigentlich müssten die zwischen Minuten bereits durch das `group by tim (1h)` ausgefiltert werden. Sofern nicht gerade um den Stundenwechsel herum viele Werte fehlen.

endurance1968, 2025-01-31:
>Kann es sein, dass bei Dir EVCC über die Stundengrenze hinweg down war und Du dann danach einen Restart gemacht hast?
Nein die Stundenwerte sind ja auch alle da. Sind nur vom hochfahren immer je ein zusätzlicher drin.

>SELECT last("value") * 100 FROM "tariffGrid" GROUP BY time(1h)

das sieht besser aus. bei mir orignal
SELECT "value" * 100 FROM "tariffGrid" WHERE $timeFilter tz('$timezone')

cschlipf, 2025-01-31:
In den Dashboard settings kannst Du die Dashboard Version sehen:

[image]

Bringt bei Library Panels leider nicht so viel :-(. Hilft dann nur, wenn man sichergestellt hat, dass die Libraries exakt zum Dashboard passen - das Problem kennst Du ja schon ;-)

closed 2025-01-31 by endurance1968