ha-puzzles/evcc-grafana-dashboards #87: Minor: Eigenverbrauch fehlt im Monatsdashboard wenn keine eingespeiste Energie an dem Tag

endurance1968, 2025-02-05:
[image]

Im Bild müsste am 5 und 18.1 der Eigenverbrauch auf 100% stehen, da keine Energie eingespeist wurde. Wenn es aber keine eingespeiste Energie gibt gibt es den Wert in der Timeseries nicht und als Folge wird auch kein Eigenverbrauch berechnet. 
Keine Ahnung ob man das Lösen kann.

cschlipf, 2025-02-05:
Ich schau mal rein.

Zum Jahresvergleich: Ich habe da einige Beta Features von Grafana 11.5 genutzt. Ich vermute die Fehler kommen davon? Welche Grafana Version hast Du?

endurance1968, 2025-02-05:
Hab noch Grafana 11.4.0 - kann aber mal updaten

done nun  11.5.1 und besser

[image]

cschlipf, 2025-02-05:
Danke! Gut zu wissen für die Release Notes.

cschlipf, 2025-02-05:
Zurück zum eigentlichen Problem. Am 5.1. und am 18.1. meintest Du da wäre nichts eingespeist worden. Ich vermute mal für die Tage steht da in der Datenbank gar nichts drin? Daher kann dann für diese Tage die Expression nichts zurückliefern.

Kannst Du mal in der influx folgendes machen?

```sql
precision rfc3339
use evcc_aggr
select * from feedInDailyEnergy where time > '2025-01-01T00:00:00.000000000Z' AND time < '2025-01-31T23:59:00.000000000Z'
```

Und das Ergebnis hier posten?

endurance1968, 2025-02-06:
```
> use evcc_agg
Using database evcc_agg
> select * from feedInDailyEnergy where time > '2025-01-01T00:00:00.000000000Z' AND time < '2025-01-31T23:59:00.000000000Z'
name: feedInDailyEnergy
time                 day month value              year
----                 --- ----- -----              ----
2025-01-01T23:00:00Z 02  01    642.8861111111111  2025
2025-01-02T23:00:00Z 03  01    186.3138888888889  2025
**2025-01-03T23:00:00Z 04  01    390.18611111111113 2025
2025-01-05T23:00:00Z 06  01    388.9111111111111  2025**
2025-01-06T23:00:00Z 07  01    315.28333333333336 2025
2025-01-07T23:00:00Z 08  01    731.3194444444445  2025
2025-01-08T23:00:00Z 09  01    234.50555555555556 2025
2025-01-09T23:00:00Z 10  01    263.64166666666665 2025
2025-01-10T23:00:00Z 11  01    75.30833333333334  2025
2025-01-11T23:00:00Z 12  01    226.875            2025
2025-01-12T23:00:00Z 13  01    96.69166666666666  2025
2025-01-13T23:00:00Z 14  01    460.925            2025
2025-01-14T23:00:00Z 15  01    649.6333333333333  2025
2025-01-15T23:00:00Z 16  01    109.21111111111111 2025
**2025-01-16T23:00:00Z 17  01    54.2               2025
2025-01-18T23:00:00Z 19  01    142.8824074074074  2025**
2025-01-19T23:00:00Z 20  01    595.2722222222222  2025
2025-01-20T23:00:00Z 21  01    608.1944444444445  2025
2025-01-21T23:00:00Z 22  01    372.2027777777778  2025
2025-01-22T23:00:00Z 23  01    66.11666666666666  2025
2025-01-23T23:00:00Z 24  01    185.53055555555557 2025
2025-01-24T23:00:00Z 25  01    309.7083333333333  2025
2025-01-25T23:00:00Z 26  01    56.63055555555555  2025
2025-01-26T23:00:00Z 27  01    101.58055555555555 2025
2025-01-27T23:00:00Z 28  01    733.7111111111111  2025
2025-01-28T23:00:00Z 29  01    589.05             2025
2025-01-29T23:00:00Z 30  01    609.1111111111111  2025
2025-01-30T23:00:00Z 31  01    477.7611111111111  2025
2025-01-31T23:00:00Z 01  02    287.15277777777777 2025
```

[image]

cschlipf, 2025-02-06:
Danke! Das bestätigt meine Vermutung. Kommt bei mir nicht vor, da bei mir irgendwie immer ein paar W eingespeist werden. Nun habe ich mal in die Queries reingeschaut - das was ich vermutet hatte, was es leicht fixen könnte ist bei dieser Query im Aggregationsscript schon drin (`fill(0)`). Ich muss da mal tiefer reinschauen.

cschlipf, 2025-02-06:
Kannst Du mal folgendes ausführen:

```bash
evcc-influx-aggregate.sh --day 2025 1 5
```

Ich vermute da wird folgende Meldung kommen:

```
There is no data from gridPower for feedInDailyEnergy. This may be fine, if no data had been collected for this day and this measurement.
```

Korrekt?

cschlipf, 2025-02-06:
Merci. Wie vermutet. Ich bin dran.

closed 2025-02-06 by cschlipf

cschlipf, 2025-02-06:
Vielen Dank für's testen.

Du kannst jederzeit die Timezone Variable überschreiben. Mit ist bewusst, dass das nicht immer funktioniert - aber ich wollte einen default, der bei manchen eventuell eine Konfiguration überflüssig macht.

endurance1968, 2025-02-07:
>Dokumentiert ist das aber hier unter Punkt 8: https://github.com/ha-puzzles/evcc-grafana-dashboards/tree/main/scripts

Ah OK - ich hab nur die Hauptseite nach Zeitzone durchsucht ob das irgendwo dokumentiert ist. die doku zum script hab ich ehrlich gesagt noch nie angeschaut :(

PR #88 (2025-02-06): Fix #87 Fehlender Eigenverbrauch
Closes #87

Wenn in der Datenbank keine Energien sind, sollte das Aggregationsscript dann 0 für die Energie schreiben, damit keine darauf abhängigen Berechnungen fehl schlagen.
commit 75d4568f (2025-02-06): Fix #87 Fehlender Eigenverbrauch (#88)

Closes #87

Wenn in der Datenbank keine Energien sind, sollte das Aggregationsscript dann 0 für die Energie schreiben, damit keine darauf abhängigen Berechnungen fehl schlagen.