ha-puzzles/evcc-grafana-dashboards #91: Improvement: Konsistente Eingabe der Variablen beim Import

endurance1968, 2025-02-08:
Moin,

Du hast angefangen die Variablen schon beim Import anpassen zu können.
Das finde ich super, da hier das Autofill vom Browser funktioniert und man sich Copy Past oder Tipparbeit spart.

Das ist aber noch nicht konsequent für alle Variablen und Dashboards durchgezogen. Wäre noch superer :) wenn man überall schon beim Import die Variablen ändern könnte.

cschlipf, 2025-02-08:
Hoppla. Da habe ich gar nichts gemacht. Scheint ein neues Feature von Grafana zu sein.

Habe schone eine Vermutung. Ich habe angefangen versteckte Variablen zu nehmen - dann aber bemerkt, dass es auch Konstanten gibt. War dann aber zu faul die ganzen Variablen nochmal neu als Konstanten anzulegen. Ich vermute die Konstanten werden beim Import abgefragt - die Variablen nicht.

Zeit mal ne Test Instanz für Grafana aufzusetzen.

cschlipf, 2025-02-08:
Wie vermutet. Konstanten werden abgefragt, Variablen nicht. Macht viel Sinn nun konsequent auf Konstanten zu setzen.

cschlipf, 2025-02-08:
Habe dabei gleich noch bemerkt, dass mit Grafana 11.5 nun Labels für Konstanten dazu gekommen sind. Gefällt mir jetzt viel besser beim Import. Doku wurde auch angepasst.

Vielen Dank für den Hinweis!

Ich warte noch darauf, dass EVCC einen neuen Release macht um die direkten Verlinkung auf Loadpoints mit dem nächsten Release zu veröffentlichen.

PR #93 (2025-02-08): Fix #91: Konsistente Nutzung von Konstanten
Closes #91

- Konsistente Nutzung von Konstanten, die beim Import des Dashboards abgefragt werden.
- Labels für Konstanten vergeben (neue Grafana Funktion)
- Update der Dokumentation.
commit f5ce96d8 (2025-02-08): Fix #91: Konsistente Nutzung von Konstanten (#93)

Closes #91
closed 2025-02-08 by cschlipf