ha-puzzles/evcc-grafana-dashboards #98: Support parameterized amounts of loadpoints and vehicles

cschlipf, 2025-02-16:
[image]

Parameterize vehicle and loadpoints through grouped queries

endurance1968, 2025-02-18:
In case the number of loadpoint will be configurable you might think of a loadpoint type. As EVCC supports heaters as well not every lp might be a charging loadpoint. No problem if the heater is the last in the list, but how to handle if the the second is a heater and the third is a charger that you would like to handle. Hmm might be no issue at all if always the name tag can be used.

endurance1968, 2025-02-18:
bin dabei

nikodeluxe, 2025-02-18:
root@influxdb:/home/evcc# ./evcc-influx-aggregate_new.sh --detect
[INFO]    [2025-02-18 22:09:46] Starting aggregation...
[INFO]    Detected vehicle 1: 
[INFO]    Detected loadpoint 1: Entfeuchter
[INFO]    Detected loadpoint 2: Heizstab

PR #101 (2025-02-18): Fix #97: Detect and aggregate all vehicles and loadpoints
Closes #97
Closes #98

- Aggregation script detects all loadpoints and vehicles from database
- Aggregation script creates aggregation for all detected elements

Requires cleanup of aggregration database and re-run.
commit f83162cc (2025-02-18): Fix #97: Detect and aggregate all vehicles and loadpoints (#101)

* Fix #97: Detect and aggregate all vehicles and loadpoints

Closes #97

- Aggregation script detects all loadpoints and vehicles from database
- Aggregation script creates aggregation for all detected elements

Requires cleanup of aggregration database and re-run.

* Updated dashboards to work with any amount of LPs and vehicles

Closes #98
closed 2025-02-18 by cschlipf