ha-puzzles/evcc-grafana-dashboards #128: 0.5.3 issues

endurance1968, 2025-03-02:
Not referenced Variables could be removed

In Daily Dashboard
[image]

cschlipf, 2025-03-02:
Not referenced Variables - wird falsch von Grafana reported. Beide Variablen werden in Library Panels referenziert.

MAX Power Threshold: Ja, leider erlaubt Grafana hier nur zahlen, keine Variablen.

Monatsdashboard ist nir bekannt. Noch keine Lösung.

Bei der Batterie. Werden hier exakt dieselben Register ausgelesen und an derselben Stelle gemessen? Das macht einen großen Unterschied.

endurance1968, 2025-03-02:
Also EVCC bekommt PV, Batterie und grid von Victron. Victron bekommt nur die Ladepunkt von EVCC.
Von daher sollten es dieselben Messpunkte sein. Ob EVCC die Werte richtig hat keine Ahnung, oder kann man Batterieladung/entladung auch irgendwo direkt in EVCC sehen?

```
meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.1.71
  port: 502
  name: victron-grid
  meterid: 40
- type: template
  template: victron-energy
  usage: pv
  host: 192.168.1.71
  port: 502
  name: victron-pv
- type: template
  template: victron-energy
  usage: battery
  host: 192.168.1.71
  port: 502
  name: victron-battery
  minsoc: 15
  maxsoc: 85
```

cschlipf, 2025-03-02:
Ich kenne das Template nicht, ob hier nur Leistung oder auch Energie übertragen wird. Energie ist optional und wird nicht immer übertragen. Bei meinem Wechselrichter zum Beispiel wird Energie zwar übertragen, aber was da herauskommt macht überhaupt keinen Sinn. Daher berechne ich die Energie durch Integrale.

Bei meinem Wechselrichter passt das mit dem was mir die Wechselrichter App anzeigt so in etwa - aber auch nicht wirklich perfekt. Nun ist aber die Frage wo der Wechselrichter misst? Mit exakt denselben Leistungswerten, die er extern herausgibt? Wie berechnet er die Leistung daraus? Viele Wechselrichter machen da keine Wissenschaft drauf und zählen einfach nur den durchschnitt mal Zeiteinheit zu einem Wert dazu. Desto kleiner die Zeiteinheit, desto genauer. Nun liefert EVCC die Leistung schon nur in Intervallen von mehreren Sekunden (default: 30s). Daher siehst Du schon, dass man gar nicht exakt genau denselben Wert erhalten kann.

76% ist für die Architektur meiner Batterie relativ normal. Falls es Dich mehr interessiert: https://www.photovoltaikforum.com/thread/238114-sofarsolar-bts-5k-speicher-erfahrungen-wirkungsgrad/?postID=4023969#post4023969

endurance1968, 2025-03-02:
Ein Verbesserungspotential habe ich evtl. noch:

[image]

Es wird das ganze Jahr 2024 als "messzeitraum" genommen. Ich habe aber nur bisserl was im Dezember drin.

Bzgl. Wirkungsgrad: bei meinem vorherigen System hatte ich einen Zweirichtungszähler davor, dann es es mehr als simpel den genauen Wirkungsgrad des gesamtsystems zu messen. Was Victron da meldet kann ich leider auch nicht sagen. Ich habe jetzt auch keinen extra Zähler mehr. 
Das die gefahrenen Leistungen einen Unterschied machen ist klar, die Grundverbräuche bleiben ja. Ich bin ja meist eher im mehre kW Bereich unterwegs.

cschlipf, 2025-03-02:
> Ein Verbesserungspotential habe ich evtl. noch:
> 
> [image]
> 
> Es wird das ganze Jahr 2024 als "messzeitraum" genommen. Ich habe aber nur bisserl was im Dezember drin.

Siehe https://github.com/ha-puzzles/evcc-grafana-dashboards/tree/main/dashboards#all-time-1, 2.ter Punkt ;-)

cschlipf, 2025-03-02:
Wegen dem Versatz im Monatsdashboard. Ich sehe ihn aktuell für den März, wo nur 2 Tage drin sind. Im Februar dagegen ist gar kein Verstatz drin. Ist das bei Dir auch so?

endurance1968, 2025-03-02:
`Wegen dem Versatz im Monatsdashboard. Ich sehe ihn aktuell für den März, wo nur 2 Tage drin sind. Im Februar dagegen ist gar kein Verstatz drin. Ist das bei Dir auch so?`

Je mehr Tage desto kleiner der Versatz. Gestern (mit nur einem Tag) war es extrem zu sehen.

endurance1968, 2025-03-02:
Ah ich spiele gerade mit dem aggregation script wegen der allteim auswertung:

`root@lx-influxdbv1:~# ./evcc-influx-aggregate.sh --month 2025 01
[DEBUG]   Aggregating month 2025-01
[INFO]    Detected vehicle 1: Altherma-3
[INFO]    Detected vehicle 2: BMW i3
[INFO]    Detected vehicle 3: Schneeflittchen
[DEBUG]   Detected 3 vehicles: Schneeflittchen BMW i3 Altherma-3
[INFO]    Detected loadpoint 1: Carport_Ecke
[INFO]    Detected loadpoint 2: Carport_Treppe
[INFO]    Detected loadpoint 3: Daikin-WP
[DEBUG]   Detected 3 vehicles: Daikin-WP Carport_Treppe Carport_Ecke
[INFO]    [2025-03-02 18:32:47] Starting aggregation...
./evcc-influx-aggregate.sh: line 498: ((: day<=: syntax error: operand expected (error token is "<=")
[INFO]    Aggregating monthly metrics for 2025-01
date: invalid date '2025-01-00T23:59:59'
[DEBUG]   2025-01: Creating monthly aggregation from pvDailyEnergy into pvMonthlyEnergy
[DEBUG]   Month has  days.
[DEBUG]   Query: SELECT sum("value") FROM pvDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
[DEBUG]   Query result (last row):
[INFO]    There is no data from pvDailyEnergy for pvMonthlyEnergy. Writing 0 as data point for this month.
[DEBUG]   Insert statement: INSERT pvMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
date: invalid date '2025-01-00T23:59:59'
[DEBUG]   2025-01: Creating monthly aggregation from homeDailyEnergy into homeMonthlyEnergy
[DEBUG]   Month has  days.
[DEBUG]   Query: SELECT sum("value") FROM homeDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
[DEBUG]   Query result (last row):
[INFO]    There is no data from homeDailyEnergy for homeMonthlyEnergy. Writing 0 as data point for this month.
[DEBUG]   Insert statement: INSERT homeMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
date: invalid date '2025-01-00T23:59:59'
[DEBUG]   2025-01: Creating monthly aggregation from gridDailyEnergy into gridMonthlyEnergy
[DEBUG]   Month has  days.
[DEBUG]   Query: SELECT sum("value") FROM gridDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
[DEBUG]   Query result (last row):
[INFO]    There is no data from gridDailyEnergy for gridMonthlyEnergy. Writing 0 as data point for this month.
[DEBUG]   Insert statement: INSERT gridMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
date: invalid date '2025-01-00T23:59:59'
`

cschlipf, 2025-03-02:
Fürchte da kann ich nicht viel machen. Mit dem Support für viele Fahrzeuge musst ich die energieLoadpoints Query auf eine Summenbildung in der Query umstellen. Hier geht die Zeit der Aggegrationen verloren und um überhaupt noch einen Zeitstempel zu haben, musste ich ein Grouping nach 1 Tag machen. Hier scheint es ein Problem zu geben, das ich aber nicht finden kann. Für die anderen Queries erhalte ich für jeden Einzelnen Tag einen Wert - auch für die zukünftigen Tage, für die es noch keine Werte gibt (musst ich so ändern, weil Du vorher lücken in gehabt hast - jetzt wird aktiv null eingeschrieben für Tage wo noch keine Werte gemessen worden sind). Für die Summen query aber eben nur für die Tage, für die es eine Summe gibt. Und Grafana scheint hier bei der Darstellung ein Problem zu haben und skaliert die Reihen mit Werten für jeden Tag etwas anders als die Reihen mit nur wenigen Werten. Deswegen gleicht sich das dann auch später wieder aus.

cschlipf, 2025-03-02:
> Ah ich spiele gerade mit dem aggregation script wegen der allteim Auswertung - die monats Aggregation scheint nicht zu funktionieren:
> 
> Ah Fehler gefunden schreibt man 2025 01 geht es nicht, aber 2025 1 geht.
> 
> > root@lx-influxdbv1:~# ./evcc-influx-aggregate.sh --month 2025 01
> > [DEBUG]   Aggregating month 2025-01
> > [INFO]    Detected vehicle 1: Altherma-3
> > [INFO]    Detected vehicle 2: BMW i3
> > [INFO]    Detected vehicle 3: Schneeflittchen
> > [DEBUG]   Detected 3 vehicles: Schneeflittchen BMW i3 Altherma-3
> > [INFO]    Detected loadpoint 1: Carport_Ecke
> > [INFO]    Detected loadpoint 2: Carport_Treppe
> > [INFO]    Detected loadpoint 3: Daikin-WP
> > [DEBUG]   Detected 3 vehicles: Daikin-WP Carport_Treppe Carport_Ecke
> > [INFO]    [2025-03-02 18:32:47] Starting aggregation...
> > **./evcc-influx-aggregate.sh: line 498: ((: day<=: syntax error: operand expected (error token is "<=")**
> > [INFO]    Aggregating monthly metrics for 2025-01
> > **date: invalid date '2025-01-00T23:59:59'**
> > [DEBUG]   2025-01: Creating monthly aggregation from pvDailyEnergy into pvMonthlyEnergy
> > [DEBUG]   Month has  days.
> > [DEBUG]   Query: SELECT sum("value") FROM pvDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
> > [DEBUG]   Query result (last row):
> > [INFO]    There is no data from pvDailyEnergy for pvMonthlyEnergy. Writing 0 as data point for this month.
> > [DEBUG]   Insert statement: INSERT pvMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
> > date: invalid date '2025-01-00T23:59:59'
> > [DEBUG]   2025-01: Creating monthly aggregation from homeDailyEnergy into homeMonthlyEnergy
> > [DEBUG]   Month has  days.
> > [DEBUG]   Query: SELECT sum("value") FROM homeDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
> > [DEBUG]   Query result (last row):
> > [INFO]    There is no data from homeDailyEnergy for homeMonthlyEnergy. Writing 0 as data point for this month.
> > [DEBUG]   Insert statement: INSERT homeMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
> > date: invalid date '2025-01-00T23:59:59'
> > [DEBUG]   2025-01: Creating monthly aggregation from gridDailyEnergy into gridMonthlyEnergy
> > [DEBUG]   Month has  days.
> > [DEBUG]   Query: SELECT sum("value") FROM gridDailyEnergy WHERE time >= '2024-12-31T23:00:00Z' AND time <= '2025-03-02T00:00:00Z' tz('Europe/Berlin')
> > [DEBUG]   Query result (last row):
> > [INFO]    There is no data from gridDailyEnergy for gridMonthlyEnergy. Writing 0 as data point for this month.
> > [DEBUG]   Insert statement: INSERT gridMonthlyEnergy,year=2025,month=01 value=0 1735686000000000000
> > date: invalid date '2025-01-00T23:59:59'

Ja, "01" ist ein String für Bash. "1" eine Zahl.

endurance1968, 2025-03-02:
Nochmals auf die FInanzen zu kommen

so würde ich mir das vorstellen:
[image]
aber die Aggregation haut ja für das ganze Jahr 2024 Daten rein - obwohl keine Quelldaten da sind.
[image]
in der ersten Grafik habe ich nur die Tage aggregiert die auch vorhanden waren. Ist bei mir easy sind nur 3. Statt einfach die Aggration für das ganze Jahre aufzurufen.
D.h. das Problem trifft nur Leute die am Anfang des Jahres fehlendes Daten zum aggregieren haben aber einfach die Aggregation des ganzen Jahres aufrufen. D.h. entweder in der Doku nochmals extra vermerken (wenn es nicht schon irgendwo steht). Oder aber das skript schlauer machen und vorher daten existenz prüfen bevor etwas in die Aggr DB geschrieben wird.

endurance1968, 2025-03-02:
Und dann ist mir gerade noch aufgefallen:

[image]

cschlipf, 2025-03-02:
Danke, Siehe #129

cschlipf, 2025-03-02:
> Nochmals auf die FInanzen zu kommen
> 
> so würde ich mir das vorstellen: [image] aber die Aggregation haut ja für das ganze Jahr 2024 Daten rein - obwohl keine Quelldaten da sind. [image] in der ersten Grafik habe ich nur die Tage aggregiert die auch vorhanden waren. Ist bei mir easy sind nur 3 Tage in 2024. Statt einfach die Aggration für das ganze Jahre aufzurufen. D.h. das Problem trifft nur Leute die am Anfang des Jahres fehlendes Daten zum aggregieren haben aber einfach die Aggregation des ganzen Jahres (gilt vermutlich auch für Monate wenn auch weniger heftig) aufrufen. D.h. entweder in der Doku nochmals extra vermerken (wenn es nicht schon irgendwo steht). Oder aber das skript schlauer machen und vorher daten existenz prüfen bevor etwas in die Aggr DB geschrieben wird.
> 
> 20 oder 280 Jahre Amortisationszeit macht halt einen kleinen Unterschied :) Ich weiss ja jetzt das ich nicht 2024 komplett aufrufen darf, also für mich auch so OK - aber falls Dir mal langweilig ist

Siehe https://github.com/ha-puzzles/evcc-grafana-dashboards/tree/main/dashboards#all-time-1, 2.ter Punkt. Du musst hier noch den Zeitraum einmal anpassen.

Bei Dir passt das nun gar nicht. Du hast nur Daten aus Wintermonaten. HIer musst Du erst noch ein Jahr abwarten, bis sich das einspielt.

endurance1968, 2025-03-02:
Das die Ergebnisdaten so nicht der Realistät entsprechen ist mir klar. Mir ging es drum, wenn man die "null" Daten gar nicht erst in die AGGR-DB schiebt, stimmt der ausgewerte Zeitraum auch ohne, dass man im Dashboard den Zeitraum anpassen muss. Das Ergebnis ist aber dasselbe.

cschlipf, 2025-03-02:
Ja, aber erinnerst Du Dich an den Fehler mit den falschen Eigenvebrauchswerten? Um den zu fixen, muss ich null reinschreiben, wenn man sonst keine Daten hat.

Entweder oder. Ich kann nicht raten, wenn ich 0 brauche, weil nur keine Daten reingekommen sind, oder wenn ich keine 0 schreiben darf, weil es nun für den Tag noch nie Daten gab.

Hier ist die Lösung den Zeitraum richtig zu setzen denke ich eine akzeptable Lösung.

endurance1968, 2025-03-03:
>Entweder oder.

Nicht ganz, es geht hier ja nicht um Tage irgendwo mittendrin. Es geht in dem Fall ja um das _Startdatum der Datenerfassung_. Es müsste doch herauszubekommen sein, wann zum erstenmal Daten in die evcc-DB geschrieben wurden - oder man gibt das als Konfigparameter an, der dann bei der Jahresweisenaggregation berücksichtig wird.

>Hier ist die Lösung den Zeitraum richtig zu setzen denke ich eine akzeptable Lösung.

Ich habe ja schon gesagt, für mich mittlerweile egal. Und ja es ist akzeptabel.

cschlipf, 2025-03-14:
Ist hier eigentlich noch ein Problem offen oder können wir das schließen?

Es ist immer etwas schwer, wenn ein Issue mit vielen Problemen aufgemacht wird. Ich kann nur eins nach dem anderen angehen. Dann besser eine Diskussion woraus man dann die einzelnen Probleme ableitet.

endurance1968, 2025-03-14:
Nur wenn Du freiwillig ein "Frühestes Startdatum" für die Aggregierung einbaust. Sprich wenn man das auf 3.12.24 setzt und dann eine Jahresaggregierung von 2024 macht nur 3.12-31.12.2024 genommen wird.

cschlipf, 2025-03-14:
Siehe Enhancement #134

closed 2025-03-14 by cschlipf