ha-puzzles/evcc-grafana-dashboards #134: Startdatum für Aggegration

cschlipf, 2025-03-14:
> Nur wenn Du _freiwillig_ ein "frühestes Startdatum" für die Aggregierung einbaust. Sprich wenn man das auf 3.12.24 setzt und dann eine Jahresaggregierung von 2024 macht nur 3.12-31.12.2024 genommen wird. 
> ist eh alles freiwillig aber Du weisst sicher was ich meine. Ich habe halt ungern unnötige Daten drin - ist bei mir halt auch der Extremfall weil ich erst am 2x.12.2024 mit evcc angefangen habe. 

 _Originally posted by @endurance1968 in [#128](https://github.com/ha-puzzles/evcc-grafana-dashboards/issues/128#issuecomment-2725426270)_

cschlipf, 2025-03-14:
Neue CLI param `--from YY MM DD --to YY MM DD`

endurance1968, 2025-03-14:
Werde ich testen - ich nehme an das funktioniert über Jahresgrenzen hinweg, sprich beliebig langer Zeitraum?

cschlipf, 2025-03-14:
Das wäre der plan. Ich verspreche aber noch nichts. Ist ehrlich gesagt in meiner Prio recht weit hinten, denn funktionieren tut es ja auch heute. Und bei Dir geht es ja nur um wenige Tage, denn statt das Jahr 2024 zu aggregieren kannst Du heute ja auch nur den Monat Dezember von 2024 aggregieren. Es geht dann nur um den 1. und 2.12 bei Dir, die zuviel sind.

endurance1968, 2025-03-15:
Yepp aber mit Deinem Vorschlag kann man immerhin auch mehrere Jahre in einem Rutsch erledigen. Im Prinzip könnte es die ganzen anderen optionen für Zeiträume ersetzen.

PR #162 (2025-04-06): Fix#134 Allow to aggregate from a given date to an optional end date (or today
Close #134
closed 2025-04-06 by cschlipf