ha-puzzles/evcc-grafana-dashboards #172: Link auf Detailed daily im Daily Dashboard

endurance1968, 2025-04-11:
Funktioniert nicht
von
https://grafana.lab.tech-freak.net/d/7Ou-Y1Rgk/evcc3a-today?orgId=1&from=now%2Fd&to=now%2Fd&timezone=browser&refresh=10s
auf
https://grafana.lab.tech-freak.net/pv/d/fed98n901h0xsb/evcc3a-today-details?from=now%2Fd&to=now%2Fd&timezone=Europe%2FBerlin

Da ist ein pv nach dem hostnamen zuviel drin?

cschlipf, 2025-04-11:
Ah, das könnte daran liegen, weil ich meine Dashboards alle in einem Ordner 'pv' abgelegt habe. Muss mal schauen ob das bei mir auch ohne Ordnernamen geht.

Vielen Dank, dass Du wieder so fix mit Testen bist.

endurance1968, 2025-04-11:
> Ah, das könnte daran liegen, weil ich meine Dashboards alle in einem Ordner 'pv' abgelegt habe. Muss mal schauen ob das bei mir auch ohne Ordnernamen geht.

also ich hab alle in einem Ordner EVCC und es geht ohne Ordnernamen

cschlipf, 2025-04-11:
> > Ah, das könnte daran liegen, weil ich meine Dashboards alle in einem Ordner 'pv' abgelegt habe. Muss mal schauen ob das bei mir auch ohne Ordnernamen geht.
> 
> also ich hab alle in einem Ordner EVCC und es geht ohne Ordnernamen

Danke! Werde ich ändern. Mein Ordner Name sollte selbstverständlich nicht vorausgesetzt werden.

commit 7bea5496 (2025-04-11): Fixed bad links (#172, #173) (#176)

Closes #172
Closes #173
closed 2025-04-11 by cschlipf