ha-puzzles/evcc-grafana-dashboards #174: Stromkosten monthly und daily dashboard stimmen nicht überein

endurance1968, 2025-04-11:
Beispiel bei mit 6.4 Monthly (beispielhaftes Datum wo es bei mir extrem auseinander lief)
[image]
Daily
[image]

cschlipf, 2025-04-11:
Hm, das ist etwas merkwürdig. Bei mir stimmen die Preise auf den ct genau überein:

[image]

cschlipf, 2025-04-11:
Ich sehe einen Unterschied in den Queries... im Details dashboard habe ich Tariff Price Interval noch nicht als Variable, sondern fest auf 1h eingestellt. Das muss ich hier noch ändern. Aber ich vermute du hast in Deiner evcc-influx-aggregate.conf für TARIFF_PRICE_INTERVAL auch "1h" stehen?

cschlipf, 2025-04-11:
Hm, ist jetzt schwer zu fixen, wenn ich es nicht reproduzieren kann :-(

Könnte mir noch vorstellen, dass bei der Aggregierung was schief gegangen ist. Kannst Du mal `evcc-influx-aggregate.sh --debug -day 2025 4 6` ausführen und mir das Ergebnis hier posten?

endurance1968, 2025-04-11:
Es kann dann doch nur am Price in der Stunde liegen. Ich hab da mal einen Debug eingebaut und der Price ist konstant 0.26 es sei denn ich mach etwas beim debug falsch

while read -r value; do
        if [ "$value" != "" ]; then
            if [ "${prices[$i]}" == "" ]; then
                logDebug "Price for index $row is empty. Using last known price $defaultPrice."
                totalPrice=$(echo "$totalPrice + $value * $defaultPrice" | bc)
            else
logDebug "Price for index $row ${prices[$i]}."
                totalPrice=$(echo "$totalPrice + $value * ${prices[$i]}" | bc)
            fi
            row=$((row+1))
        fi
ergibt
```
[DEBUG]   Price for index 0 0.2605.
[DEBUG]   Price for index 1 0.2605.
[DEBUG]   Price for index 2 0.2605.
[DEBUG]   Price for index 3 0.2605.
[DEBUG]   Price for index 4 0.2605.
[DEBUG]   Price for index 5 0.2605.
[DEBUG]   Price for index 6 0.2605.
[DEBUG]   Price for index 7 0.2605.
[DEBUG]   Price for index 8 0.2605.
[DEBUG]   Price for index 9 0.2605.
[DEBUG]   Price for index 10 0.2605.
[DEBUG]   Price for index 11 0.2605.
[DEBUG]   Price for index 12 0.2605.
[DEBUG]   Price for index 13 0.2605.
[DEBUG]   Price for index 14 0.2605.
[DEBUG]   Price for index 15 0.2605.
[DEBUG]   Price for index 16 0.2605.
[DEBUG]   Price for index 17 0.2605.
[DEBUG]   Price for index 18 0.2605.
[DEBUG]   Price for index 19 0.2605.
[DEBUG]   Price for index 20 0.2605.
[DEBUG]   Price for index 21 0.2605.
[DEBUG]   Price for index 22 0.2605.
[DEBUG]   Price for index 23 0.2605.
```
Sollte aber dem hier entsprechen oder?

[image]

`root@lx-influxdbv1:~# ./evcc-influx-aggregate.sh --debug --day 2025 4 6
[DEBUG]   Enabling debug.
[DEBUG]   Aggregating day 2025-4-6
[INFO]    Detected vehicle 1: Altherma-3
[INFO]    Detected vehicle 2: BMW i3
[INFO]    Detected vehicle 3: Schneeflittchen
[DEBUG]   Detected 3 vehicles: Schneeflittchen BMW i3 Altherma-3
[INFO]    Detected loadpoint 1: Carport_Ecke
[INFO]    Detected loadpoint 2: Carport_Treppe
[INFO]    Detected loadpoint 3: Daikin-WP
[DEBUG]   Detected 3 loadpoints: Daikin-WP Carport_Treppe Carport_Ecke
[INFO]    [2025-04-11 20:52:43] Starting aggregation...
[INFO]    Aggregating daily metrics for 2025-04-06
[DEBUG]   2025-04-06: Aggregating energy of positive values from pvPower into pvDailyEnergy
[DEBUG]   Query: SELECT integral("subquery") / 3600 FROM (SELECT mean("value") AS "subquery" FROM "pvPower" WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' AND value < 40000 AND (id::tag = '') AND "value" >=0 GROUP BY time(20s) fill(0)) WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' GROUP BY time(1d) fill(0) tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-06T00:00:00+02:00 63365.63888888889
[DEBUG]   Insert statement: INSERT pvDailyEnergy,year=2025,month=04,day=06 value=63365.63888888889 1743890400000000000
[DEBUG]   2025-04-06: Aggregating energy of positive values from homePower into homeDailyEnergy
[DEBUG]   Query: SELECT integral("subquery") / 3600 FROM (SELECT mean("value") AS "subquery" FROM "homePower" WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' AND value < 40000 AND "value" >=0 GROUP BY time(20s) fill(0)) WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' GROUP BY time(1d) fill(0) tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-06T00:00:00+02:00 73884.88587997688
[DEBUG]   Insert statement: INSERT homeDailyEnergy,year=2025,month=04,day=06 value=73884.88587997688 1743890400000000000
[DEBUG]   2025-04-06: Aggregating energy of positive values from gridPower into gridDailyEnergy
[DEBUG]   Query: SELECT integral("subquery") / 3600 FROM (SELECT mean("value") AS "subquery" FROM "gridPower" WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' AND value < 40000 AND "value" >=0 GROUP BY time(20s) fill(0)) WHERE time >= '2025-04-05T22:00:00Z' AND time <= '2025-04-06T21:59:59Z' GROUP BY time(1d) fill(0) tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-06T00:00:00+02:00 47760.680555555555
[DEBUG]   Insert statement: INSERT gridDailyEnergy,year=2025,month=04,day=06 value=47760.680555555555 1743890400000000000
[DEBUG]   2025-04-06: Aggregating energy of negative values from gridPower into feedInDailyEnergy
[... 210 lines cut ...]
[DEBUG]   Month has 30 days.
[DEBUG]   Query: SELECT sum("value") FROM vehicleDailyEnergy WHERE time >= '2025-03-31T22:00:00Z' AND time <= '2025-04-30T21:59:59Z' AND "vehicle"::tag = 'Altherma-3' tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-01T00:00:00+02:00 97362.05694444444
[DEBUG]   Insert statement: INSERT vehicleMonthlyEnergy,year=2025,month=04,vehicle=Altherma-3 value=97362.05694444444 1743458400000000000
[DEBUG]   Aggregating loadpoint Daikin-WP
[DEBUG]   2025-04: Creating monthly aggregation from loadpointDailyEnergy into loadpointMonthlyEnergy
[DEBUG]   Month has 30 days.
[DEBUG]   Query: SELECT sum("value") FROM loadpointDailyEnergy WHERE time >= '2025-03-31T22:00:00Z' AND time <= '2025-04-30T21:59:59Z' AND "loadpoint"::tag = 'Daikin-WP' tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-01T00:00:00+02:00 97362.05694444444
[DEBUG]   Insert statement: INSERT loadpointMonthlyEnergy,year=2025,month=04,loadpoint=Daikin-WP value=97362.05694444444 1743458400000000000
[DEBUG]   Aggregating loadpoint Carport_Treppe
[DEBUG]   2025-04: Creating monthly aggregation from loadpointDailyEnergy into loadpointMonthlyEnergy
[... cut ...]
[DEBUG]   2025-04: Creating monthly aggregation from loadpointDailyEnergy into loadpointMonthlyEnergy
[DEBUG]   Month has 30 days.
[DEBUG]   Query: SELECT sum("value") FROM loadpointDailyEnergy WHERE time >= '2025-03-31T22:00:00Z' AND time <= '2025-04-30T21:59:59Z' AND "loadpoint"::tag = 'Carport_Ecke' tz('Europe/Berlin')
[DEBUG]   Query result (last row): 2025-04-01T00:00:00+02:00 14560.61329116037
[DEBUG]   Insert statement: INSERT loadpointMonthlyEnergy,year=2025,month=04,loadpoint=Carport_Ecke value=14560.61329116037 1743458400000000000

[INFO]    Aggregation finished after 00h 00m 02s.
`

cschlipf, 2025-04-11:
Ich sehe hier diese Preisliste:

Grid prices: 0.2605 0.2493 0.2513 0.2579 0.2599 0.2688 0.2678 0.2696 0.2264 0.1768 0.1743 0.1439 0.1053 0.0502 0.0404 0.0934 0.1541 0.1788 0.3105 0.3332 0.3348 0.3211 0.3098 0.2957
[DEBUG]   

Und verdammte Axt... ich sehe den Fehler. Ich nehme $i um den Preisauszulesen. Es muss aber $row sein.

endurance1968, 2025-04-11:
Yepp - hab ich auch nicht gesehen - scheint dann zu passen

[image]

cschlipf, 2025-04-11:
Und ich habe noch vergessen den letzten Preis zu aktualisieren... mann, mann, mann.... Ich aggregiere gerade nochmal durch und vergleiche meine Wert.

Mit den Werten zu meinen Tibberzeiten kann ich das auch nachvollziehen.

cschlipf, 2025-04-11:
Ist gefixt. Daten müssen leider nochmal neu aggregiert werden.

Und langsam wird die Aggregation gefährlich langsam. Viel mehr kann ich da nicht mehr reinpacken :(

closed 2025-04-11 by cschlipf

endurance1968, 2025-04-12:
> Ist gefixt. Daten müssen leider nochmal neu aggregiert werden.
> 
> Und langsam wird die Aggregation gefährlich langsam. Viel mehr kann ich da nicht mehr reinpacken :(

Teste ich heute noch - 1s pro Tag ungefähr bei mir. Aber die Dashboards bieten auch immer mehr nutzen. Von daher ist das denke ich gerechtfertigt. 3min für die 3.5Monate von 2025 - mich stört das nicht, läuft ja im Hintergrund, selbst wenn es eine Stunde dauert.

PR #177 (2025-04-11): Fix $174: Wrong price calculation
Fixed using the correct price for every price interval. Previously only used the first price. Also not updating last price.

Closes #174