ha-puzzles/evcc-grafana-dashboards #179: Heizlüfter im Daily Dashboard

endurance1968, 2025-04-12:
[image]

Die aktuelle Leistung entspricht meiner Wärmepumpe - aber im Kurvenchart ist es irgendetwas - passt am ehesten zum Gesamthausverbrauch aber auch nicht so richtig.

cschlipf, 2025-04-12:
Ja, das ist extPower. Das ist noch ein Experiment für #123. Sollte eigentlich ausgeblendet sein.

endurance1968, 2025-04-12:
OK - ich hab mal meine circuit und meter section oben mit reingenommen - und blende das bei mir aus.

cschlipf, 2025-04-12:
Gerade ist der noch nicht releaste Zustand im großen Umbau. Aber vielen Dank für Deine Mühe das zu testen.

endurance1968, 2025-04-12:
> Gerade ist der noch nicht releaste Zustand im großen Umbau. Aber vielen Dank für Deine Mühe das zu testen.

Hatte gerade Zeit und Interesse - reiner Eigennutz. Die nächsten Wochen werde ich weniger Zeit haben und Doch nicht mehr so fluten.

PR #182 (2025-04-12): Fix #179: Hide ext devices in today
Closes #179
commit 31d6b647 (2025-04-12): Fix #179: Hide ext devices in today (#182)

Closes #179
closed 2025-04-12 by cschlipf