ha-puzzles/evcc-grafana-dashboards #185: Name numbered, unnamed entities (PV, ext,...)

cschlipf, 2025-04-13:
Assign names to unamed entities that are only identified by index numbers, like PV or ext (see #123). Also use in Details dashboard:

[image]

Idea: Let the option `--detect` detect the used index number. Then use a new option `--assign-names` to assign names in an interactive CLI flow that asks for names for which index, which name should be given. This will be stored as a measurement in the Influx aggregation DB.

endurance1968, 2025-04-13:
OT:
Wo ich mir das gerade so anschaue - hat es irgendeinen Vorteil für EVCC, EVCC die einzelnen PV Anlagen wissen zu lassen (statt nur die Gesamtsumme)? 
Bisher habe ich nur die Gesamtanlage an EVCC weiter geleitet (das summiert Victron schon zusammen).
Ich nehme an, dass macht nur bei der Auswertung dann einen Unterschied oder?

cschlipf, 2025-04-13:
Genau. EVCC ist es egal. Ich habe die PV Meter so konfiguriert, dass ich meine beiden Strings einzeln sehe. Das EVCC Template summiert diese hier aber auch auf. Es gibt nun ja aber auch noch Leute, die mehrere Wechselrichter haben und diese getrennt monitoren wollen.

Und neben der PV haben wir dasselbe Spiel bei den ext Meters, wenn man einzelne, nicht von EVCC gesteuerte Verbraucher monitoren will (eigentlich auch etwas, das EVCC egal ist - da scheinen sie etwas inkonsequent zu sein).

endurance1968, 2025-04-13:
> Es gibt nun ja aber auch noch Leute, die mehrere Wechselrichter haben

Ja aktuell 5 bei mir - ist halt immer mal wieder etwas dazugekommen. 2-3 weitere sind dieses und nächstes Jahr geplant (sind dann aber alles Balkonkraftwerkswechselrichter) - kleine Flächen für die Abendsonne. Evtl. werde ich 1-2 der alten auch direkt als DC-DC umbauen. 
Um auf das Topic zu kommen - ohne automatische Benamsung hätte ich da keine Lust das immer manuell nachzufrickeln. Da bleibe ich bei einer Anlage im EVCC.

cschlipf, 2025-04-13:
Die Idee wäre aber ja gerade das Mapping von Nummer zu Namen ein einziges Mal in der Datenbank zu speichern und eben nicht in den Dashboards als hardcodierte Overrides, die Du jedes einzelne Mal aktualisieren musst (wie es heute ist).

endurance1968, 2025-04-13:
> Die Idee wäre aber ja gerade das Mapping von Nummer zu Namen ein ein

Ich hätte es evtl. anders herum und somit positiv formulieren sollen - wenn Du das so wie gehofft hinbekommst, dass man es nur einmal oder gar ohne manuelle Aktion... würde ich glatt alle PV Anlagen einpflegen.

cschlipf, 2025-04-13:
Ohne manuell nicht (erst eventuell wenn EVCC eine Benamung der Meters macht - aber dann musst Du es dort einmal machen).

Aber eben nur einmal (solange die Meters Configuration in EVCC nicht geändert wird)

cschlipf, 2025-05-11:
Es sieht so aus als würde EVCC die Titles von Meters nicht in Influx schreiben (getestet mit PV). Daher wird diese Erweiterung so oder so notwendig sein.

cschlipf, 2025-06-12:
Nach längerer Überlegung sehe ich hier leider doch noch einige Challenges, die hier einigen Konfigurationsaufwand aus Userseite nachziehen. Ich setze daher eher auf ein Enhancement in EVCC um dies zu lösen: https://github.com/evcc-io/evcc/issues/21805

Das wäre die sauberste Lösung und benötigt keine weitere Konfiguration.

Ich mache daher hier erstmal zu.

closed 2025-06-12 by cschlipf