ha-puzzles/evcc-grafana-dashboards #200: Show effective price in price charts

cschlipf, 2025-05-24:
Hmm almost
[image]
I would like to have a fourth line (or replacing average) for the effective daily price in the monthly price chart. THe effective price should always be between min and average otherwise something went wrong.
similar in the yearly 
[image]
Then it would be easier to see how effective the optimization algo works to reduce total costs.

_Originally posted by @endurance1968 in https://github.com/ha-puzzles/evcc-grafana-dashboards/discussions/199#discussioncomment-13254954_

cschlipf, 2025-07-13:
[image]

Tipp: Um es besser sehen zu können im Chart auf 'View' klicken:

[image]

closed 2025-07-13 by cschlipf

endurance1968, 2025-07-14:
Danke Dir. Werde ich demnächst mal wieder all charts neu importieren und testen.

commit 6fc4eb92 (2025-07-13): Fix #200: Show effective price in charts (#206)

Show effective price as comparison in monthly and yearly charts.

Closes #200