ha-puzzles/evcc-grafana-dashboards #210: Neues Dashboard: EVCC: Today - Details 0.6.1 keine Auswahlmöglichkeiten beim Import...

Breakyman, 2025-07-29:
Das neue Dashboard EVCC: Today - Details lassen sich beim Import keine Settings setzen (DB, etc...)

[image]

Breakyman, 2025-07-29:
Ja bei den anderen schon...

cschlipf, 2025-07-29:
Kannst Du mal versuchen das Dashboard von dieser Datei zu installieren?

https://github.com/ha-puzzles/evcc-grafana-dashboards/blob/main/dashboards/dashboards/EVCC_%20Today%20-%20Details.json

Eigentlich sind in der Datei noch alle Variablen drin. Ich wundere mich gerade ob da beim Upload zu Grafana.com was schief ging.

Breakyman, 2025-07-29:
Auch damit der gleiche Fehler... keine Auswahl der Datenbank

Breakyman, 2025-07-29:
Dadurch kein Inhalt im Dashboard:
[image]

cschlipf, 2025-07-29:
Oops... Ok, da muss ich genauer reinschauen. Vielen Dank.

tugsi, 2025-07-31:
Dasselbe auch bei mir:
Grafana v12.1.0 (ccd7b6ce7e)

cschlipf, 2025-08-01:
Vielen Dank Euch... hab den Fehler... Fix kommt.

closed 2025-08-01 by cschlipf

Breakyman, 2025-08-01:
Wie immer viele Dank für Deinen super Support!!!

commit c53efcf7 (2025-08-01): Fix #210: Missing parameters (#211)

Closes #210